ros2/rmw_opensplice #43: update implementation to check for error in all cases

dirk-thomas, 2015-06-23:
Similar to ros2/rmw_connext#43

wjwwood, 2015-07-02:
This should be resolved by https://github.com/ros2/rmw_opensplice/pull/45

wjwwood, 2015-07-15:
@dirk-thomas I've re-reviewed this in the light of #45 being merged and I now think it is resolved. I did quite a bit more work after we originally discussed closing this with #45. Please have a look and see what you think.

Note that I created an issue for the destructors of the requester and repliers, but that doesn't really apply to this issue since there are no actions which are not being checked for errors, but instead we're not doing some clean up that we should be doing:

https://github.com/ros2/rmw_opensplice/issues/47

dirk-thomas, 2015-07-16:
After taking a closer look to the current state I saw a few cases which are not being checked as defensively as in rmw_connext. I will create a PR adding those checks.

dirk-thomas, 2015-07-16:
Please see #48 for additional checks.

closed 2015-07-17 by dirk-thomas