ros2/rmw_opensplice #69: ROS_DOMAIN_ID is faling on opensplice

tfoote, 2015-08-18:
We have added `ROS_DOMAIN_ID` as a way to select separate domains. However it's causing opensplice to crash.  

It was reported that removing the environment variable would let it work but that does not appear to hold. 

```
tfoote@yeti:~/work/ros2/test_param Last: [0] (0s Seconds)
$ ./build/test_rclcpp/local_parameters_cpp__rmw_opensplice_cpp
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from parameters
[ RUN      ] parameters.local_synchronous
unknown file: Failure
C++ exception with description "could not create node: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232" thrown in the test body.
[  FAILED  ] parameters.local_synchronous (1 ms)
[ RUN      ] parameters.local_asynchronous
[rmw|error_handling.c:44] error string being overwritten: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232
unknown file: Failure
C++ exception with description "could not create node: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232" thrown in the test body.
[  FAILED  ] parameters.local_asynchronous (4 ms)
[----------] 2 tests from parameters (5 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (5 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] parameters.local_synchronous
[  FAILED  ] parameters.local_asynchronous

 2 FAILED TESTS
tfoote@yeti:~/work/ros2/test_param Last: [1] (0s Seconds)
$ env | grep ROS
ROS_ROOT=/unset
ROS_DOMAIN_ID=84
ROS_EMAIL=tfoote@osrfoundation.org
tfoote@yeti:~/work/ros2/test_param Last: [0] (0s Seconds)
$ unset ROS_DOMAIN_ID
tfoote@yeti:~/work/ros2/test_param Last: [0] (0s Seconds)
$ ./build/test_rclcpp/local_parameters_cpp__rmw_opensplice_cpp
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from parameters
[ RUN      ] parameters.local_synchronous
unknown file: Failure
C++ exception with description "could not create node: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232" thrown in the test body.
[  FAILED  ] parameters.local_synchronous (1 ms)
[ RUN      ] parameters.local_asynchronous
[rmw|error_handling.c:44] error string being overwritten: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232
unknown file: Failure
C++ exception with description "could not create node: failed to create domain participant, at /home/tfoote/work/ros2/test_param/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/functions.cpp:232" thrown in the test body.
[  FAILED  ] parameters.local_asynchronous (0 ms)
[----------] 2 tests from parameters (1 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (1 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] parameters.local_synchronous
[  FAILED  ] parameters.local_asynchronous

 2 FAILED TESTS
```

William opened a thread on opensplice forums: http://forums.opensplice.org/index.php?/topic/2591-programmatically-choosing-the-domain-id/?p=4469

I've seen reference to an xml file which can be set to match the ROS_DOMAIN_ID but I do not know where to find that. 

We should update https://github.com/ros2/examples/wiki to reflect this info.

dirk-thomas, 2015-08-18:
If the environment variable `ROS_DOMAIN_ID` is not set and the OpenSplice configuration is unmodified (contains domain id 0) I am pretty confident that it works. Because this is the exact setup we use for all CI jobs which obviously pass the tests.

If you set the `ROS_DOMAIN_ID` environment variable you have to make sure to update the OpenSplice configuration file accordingly. You find it by looking in your environment. It is referenced in the `OSPL_URI` environment variable. I have also confirmed locally that this combination works fine.

The wiki currently does not mention the `ROS_DOMAIN_ID` variable. While it would be a good enhancement I don't see a strong need to add it immediately.

jacquelinekay, 2015-08-18:
the default config file for deb install appears to be: /usr/etc/opensplice/config/ospl.xml

gerkey, 2015-08-18:
@tfoote to check that setting the environment variable to match the config file setting. There's some other problem that is causing failures on the farm.

tfoote, 2015-08-24:
Added to documentation at the bottom of https://github.com/ros2/examples/wiki

closed 2015-08-24 by tfoote