ros2/rmw_opensplice #76: Server receives uninitialized requests

esteve, 2015-08-25:
See ros2/rclcpp#95

* `add_two_ints_server__rmw_opensplice_cpp` / `add_two_ints_client__rmw_opensplice_cpp`: the server receives additional uninitialized requests
* `test_replier_cpp__rmw_opensplice_cpp` / `test_requester_cpp__rmw_opensplice_cpp` has the same problem - the test does not cover that (even with ros2/system_tests#34) since the client succeeds

jacquelinekay, 2015-09-10:
I haven't seen this issue yet. Is anyone else able to give more details about how to reproduce it/how to know it's happening? @dirk-thomas originally reported it, maybe he can help?

esteve, 2015-09-10:
This seems to be fixed by:

https://github.com/ros2/rmw_opensplice/pull/78

closed 2015-09-10 by jacquelinekay