ros2/rmw_opensplice #79: Fail gracefully if opensplice is not installed

scpeters, 2015-08-29:
I just tested building ros2 from source with an RTIConnext debian package installed but without opensplice installed. I used the [ros2-rc1.repos](https://gist.github.com/dirk-thomas/25906bd5af20e64d311c) install file. It failed with the following message:

~~~
+++ Building 'opensplice_cmake_module'
Running cmake because arguments have changed.
==> '. /root/ros2_ws/build/opensplice_cmake_module/cmake__build.sh && /usr/bin/cmake /root/ros2_ws/src/ros2/rmw_opensplice/opensplice_cmake_module -DAMENT_ENABLE_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/root/ros2_ws/install' in '/root/ros2_ws/build/opensplice_cmake_module'
 - Found ament_cmake: 0.0.0 (/root/ros2_ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.4", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
CMake Error at cmake/Modules/FindOpenSplice.cmake:93 (find_package):
  Could not find a package configuration file provided by "opensplice" with
  any of the following names:

    openspliceConfig.cmake
    opensplice-config.cmake

  Add the installation prefix of "opensplice" to CMAKE_PREFIX_PATH or set
  "opensplice_DIR" to a directory containing one of the above files.  If
  "opensplice" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
~~~

dirk-thomas, 2015-08-29:
The idea is to remove the package from the workspace or ignore it with a `AMENT_IGNORE` marker file if you don't want to build it.

scpeters, 2015-08-29:
@tfoote said to file this issue; shouldn't it be able to fail gracefully? The `rmw_connext` package is in the workspace too, but I didn't notice a failure when I built without RTI support.

dirk-thomas, 2015-08-31:
Addressed by #80.

commit d6b80867 (2015-08-31): fix CMake module to not require OpenSplice but fall back gracefully (fix #79)
closed 2015-08-31 by dirk-thomas