ros2/rmw_opensplice #99: Handle Server runs out of handle space

jacquelinekay, 2015-12-03:
Originally discovered while testing ros2/rclcpp#165

The easiest way to reproduce this is to choose any gtest that exercises the client library using `rmw_opensplice_cpp` with the command line argument `--gtest_repeat=-1`, which will repeat the test indefinitely.

For example: `./gtest_executor__rmw_opensplice_cpp --gtest_repeat=-1`

After over 100 to several hundred repeated iterations (depending on the machine), the process will exit with the following error:

```
========================================================================================
Report      : ERROR
Date        : Thu Dec 03 09:53:20 PST 2015
Description : The Handle Server ran out of handle space
Node        : pol
Process     : gtest_executor__rmw_opensplice_cpp <24792>
Thread      : main thread 7fde172d1780
Internals   : V6.4.140407OSS///kernel::v_handle::v_handleServerRegister/v_handle.c/235/0/1449165200.978805977
```

To me, this suggests that we need to improve our teardown of middleware handles within rmw.

(I am investigating if there is a similar problem in Connext now.)

gerkey, 2015-12-03:
I don't know whether it's relevant, but I got this error after ~100 iterations of that test in a Debug build of ROS 2.

jacquelinekay, 2015-12-03:
I'm running the same test case used in the example in the description with Connext and I'm on iteration 900 now. I don't think this is a problem in Connext, unless their resource storage is significantly better than Opensplice.

gerkey, 2016-06-07:
Possibly related are crashes that we're seeing with repeated invocations of certain tests (e.g., http://ci.ros2.org/job/ci_windows/1417/testReport/%28root%29/projectroot/gtest_executor__rmw_opensplice_cpp/). Below is a backtrace showing that a test segfaults while trying to create a node. This crash seems to go away if you comment out all the other test cases in that file. That makes me think that OpenSplice isn't being correctly cleaned up when used repeatedly in a single process.

```
(gdb) bt
#0  0x00007f12f003beb7 in mprotect () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f12f0cf453a in allocate_stack (stack=<synthetic pointer>, pdp=<synthetic pointer>, attr=<optimized out>)
    at allocatestack.c:649
#2  __pthread_create_2_1 (newthread=0x216fad0, attr=0x7fff91f63180, start_routine=0x7f12efcae4a0, arg=0x214e930)
    at pthread_create.c:539
#3  0x00007f12efcaecfa in os_threadCreate () from /usr/lib/libddskernel.so
#4  0x00007f12efc93e13 in u_participantInit () from /usr/lib/libddskernel.so
#5  0x00007f12efc93fd9 in u_participantNew () from /usr/lib/libddskernel.so
#6  0x00007f12efca4a82 in _DomainParticipantNew () from /usr/lib/libddskernel.so
#7  0x00007f12efbe2506 in gapi_domainParticipantFactory_create_participant () from /usr/lib/libddskernel.so
#8  0x00007f12ef92ee89 in DDS::DomainParticipantFactory::create_participant(int, DDS::DomainParticipantQos const&, DDS::DomainParticipantListener*, unsigned int) () from /usr/lib/libdcpssacpp.so
#9  0x00007f12f08c1fdf in rmw_create_node (name=0x215f570 "multiple_executors_1", domain_id=0)
    at /home/gerkey/ros2_ws_debug/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_node.cpp:97
#10 0x00007f12f0f0e0dd in rcl_node_init (node=0x214e4d0, name=0x215f570 "multiple_executors_1", options=0x7fff91f63a50)
    at /home/gerkey/ros2_ws_debug/src/ros2/rcl/rcl/src/rcl/node.c:109
#11 0x00007f12f142b06b in rclcpp::node::Node::Node (this=0x2186ee0, node_name="multiple_executors_1", 
    context=std::shared_ptr (count 20, weak -1) 0x7fff91f63b70, use_intra_process_comms=false)
    at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:87
#12 0x00007f12f142ab82 in rclcpp::node::Node::Node (this=0x2186ee0, node_name="multiple_executors_1", use_intra_process_comms=false)
    at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:30
#13 0x00000000004f4229 in __gnu_cxx::new_allocator<rclcpp::node::Node>::construct<rclcpp::node::Node, char const (&) [21]> (
    this=0x7fff91f63cf7, __p=0x2186ee0) at /usr/include/c++/5/ext/new_allocator.h:120
#14 0x00000000004f22e0 in std::allocator_traits<std::allocator<rclcpp::node::Node> >::construct<rclcpp::node::Node, char const (&) [21]> (__a=..., __p=0x2186ee0) at /usr/include/c++/5/bits/alloc_traits.h:530
#15 0x00000000004f0222 in std::_Sp_counted_ptr_inplace<rclcpp::node::Node, std::allocator<rclcpp::node::Node>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<char const (&) [21]> (this=0x2186ed0, __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:522
#16 0x00000000004ededf in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rclcpp::node::Node, std::allocator<rclcpp::node::Node>, char const (&) [21]> (this=0x7fff91f63fa8, __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:617
#17 0x00000000004eb6d8 in std::__shared_ptr<rclcpp::node::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::node::Node>, char const (&) [21]> (this=0x7fff91f63fa0, __tag=..., __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:1097
#18 0x00000000004e94d2 in std::shared_ptr<rclcpp::node::Node>::shared_ptr<std::allocator<rclcpp::node::Node>, char const (&) [21]> (
    this=0x7fff91f63fa0, __tag=..., __a=...) at /usr/include/c++/5/bits/shared_ptr.h:319
#19 0x00000000004e6dba in std::allocate_shared<rclcpp::node::Node, std::allocator<rclcpp::node::Node>, char const (&) [21]> (__a=...)
    at /usr/include/c++/5/bits/shared_ptr.h:620
#20 0x00000000004e33ee in std::make_shared<rclcpp::node::Node, char const (&) [21]> () at /usr/include/c++/5/bits/shared_ptr.h:636
#21 0x00000000004e10bf in rclcpp::node::Node::make_shared<char const (&) [21]> ()
    at /home/gerkey/ros2_ws_debug/install/include/rclcpp/node.hpp:59
#22 0x00000000004d074f in test_executor__rmw_opensplice_cpp_multiple_executors_Test::TestBody (this=0x2132050)
    at /home/gerkey/ros2_ws_debug/src/ros2/system_tests/test_rclcpp/test/test_executor.cpp:96
#23 0x0000000000525bd0 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x2132050, 
    method=&virtual testing::Test::TestBody(), location=0x5367fb "the test body") at /usr/src/gtest/src/gtest.cc:2078
#24 0x0000000000520dbd in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x2132050, 
    method=&virtual testing::Test::TestBody(), location=0x5367fb "the test body") at /usr/src/gtest/src/gtest.cc:2114
#25 0x00000000005068aa in testing::Test::Run (this=0x2132050) at /usr/src/gtest/src/gtest.cc:2151
#26 0x00000000005070dc in testing::TestInfo::Run (this=0x2132500) at /usr/src/gtest/src/gtest.cc:2326
#27 0x0000000000507743 in testing::TestCase::Run (this=0x21321b0) at /usr/src/gtest/src/gtest.cc:2444
#28 0x000000000050e61e in testing::internal::UnitTestImpl::RunAllTests (this=0x2131d80) at /usr/src/gtest/src/gtest.cc:4315
#29 0x0000000000526b23 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (
    object=0x2131d80, 
[... cut ...]
    method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x50e366 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x537090 "auxiliary test code (environments or event listeners)") at /usr/src/gtest/src/gtest.cc:2114
#31 0x000000000050d216 in testing::UnitTest::Run (this=0x76ef00 <testing::UnitTest::GetInstance()::instance>)
    at /usr/src/gtest/src/gtest.cc:3926
#32 0x00000000004dedf6 in RUN_ALL_TESTS () at /usr/include/gtest/gtest.h:2288
#33 0x00000000004d210a in main (argc=1, argv=0x7fff91f64638)
    at /home/gerkey/ros2_ws_debug/src/ros2/system_tests/test_rclcpp/test/test_executor.cpp:224
```

dirk-thomas, 2016-06-07:
The problem is still there for me and fails with out-of-handle-space after 166 iterations. Interestingly the first iteration only takes ~60ms but every iteration runs slower. Iteration 165 took over 2 seconds. Also the memory consumption is increasing steadily.

tfoote, 2016-06-07:
Yeah, and the thread count is steadily increasing. Running in gdb, each iteration seems to orphan several threads per unit test. https://gist.github.com/tfoote/f143656cfbeb32526f9a7e3c9fc124f8#file-output-in-gdb-showing-thread-leakage

In the 20th iteration I started getting `new_participant(3aec8887:f997:2:1c1, 1) failed: max participants reached` and `handleSubscriptions: participant not found, presumably already been deleted`

I got a slightly different behavior of hanging, which let me attach gdb: 
gdb and console output here: https://gist.github.com/tfoote/f143656cfbeb32526f9a7e3c9fc124f8

Unfortunately the backtraces are mostly opaque:
```
(gdb) bt full
#0  pthread_cond_timedwait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:238
No locals.
#1  0x00007f9d9b032c9e in os_condTimedWait () from /usr/lib/libddsos.so
No symbol table info available.
#2  0x00007f9d992eecbc in c_condTimedWait () from /usr/lib/libddsdatabase.so
No symbol table info available.
#3  0x00007f9d98e42bc6 in v_leaseManagerMain () from /usr/lib/libddskernel.so
No symbol table info available.
#4  0x00007f9d9ae081f3 in ?? () from /usr/lib/libddsuser.so
No symbol table info available.
#5  0x00007f9d9b03b7f6 in ?? () from /usr/lib/libddsos.so
No symbol table info available.
#6  0x00007f9d9c1e5184 in start_thread (arg=0x7f9c9b917700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7f9c9b917700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140310601627392, 2549356677189420281, 1, 1, 140310601628096, 140310601627392, 
                -2569280193526676231, -2568711669261843207}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, 
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#7  0x00007f9d9b5ae37d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

We're going to need some debug symbols in Opensplice... I can't find any source of precompiled debug libraries does anyone know of them?  @j-rivero could we create a libopensplice64-dbg in our build process?

tfoote, 2016-06-08:
I've dug a lot deeper into it. The main conclusion so far I've found is that we're repeatedly calling `rcl_node_init` but not `rcl_node_fini`. However I am not sure that's causing the segfault, it is just leaking resources. And the segfault can happen as early as 20 iterations in or as late as 300 iterations. It is causing a leak of threads. For each iteration of the gtest it creates 9 threads and 3 exit, leaking 6 threads. I believe that this is causing the significant slow down in performance. It's not unusual to have thousands of threads in gdb once I get a backtrace. 

The segfault is occurring when iterating the waitset. The `next` iterator appears to have garbage after a few links. It is always in the `c_iterTake` method inside `_ConditionRemoveWaitset` in `gapi_waitSet_detach_condition` in `DDS::WaitSet::detach_condition` in `rmw_wait`. I suspect that there's another thread that's mutating the list while it's being iterated, but I'm having trouble localizing the issue due to the leaking threads at the moment. 

There are lots of notes at: https://gist.github.com/tfoote/6dfe930f14a3c9fbe85297ea1bab32be

gerkey, 2016-06-08:
@tfoote, how did you determine that `rmw_node_fini` isn't being called? I instrumented create and destroy calls in `rcl` and `rmw_opensplice_cpp`. As far as I can see, they all line up as expected. Below is an example trace. The `return OK` line shows that we're returning from the end up `rmw_destroy_node` in `rmw_opensplice_cpp` (as opposed to bailing out early).

```
[==========] Running 4 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 4 tests from test_executor__rmw_opensplice_cpp
[ RUN      ] test_executor__rmw_opensplice_cpp.recursive_spin_call
rcl_node_init: recursive_spin_call 0x7fded33fc070
rmw_create_node: recursive_spin_call
rcl_node_fini: 0x7fded33fc070
rmw_destroy_node: 0x7fded33fba00
return OK
[       OK ] test_executor__rmw_opensplice_cpp.recursive_spin_call (26 ms)
[ RUN      ] test_executor__rmw_opensplice_cpp.multithreaded_spin_call
rcl_node_init: multithreaded_spin_call 0x7fded3ec8f50
rmw_create_node: multithreaded_spin_call
rcl_node_fini: 0x7fded3ec8f50
rmw_destroy_node: 0x7fded3eeb470
return OK
[       OK ] test_executor__rmw_opensplice_cpp.multithreaded_spin_call (27 ms)
[ RUN      ] test_executor__rmw_opensplice_cpp.multiple_executors
rcl_node_init: multiple_executors_1 0x7fded3d59ae0
rmw_create_node: multiple_executors_1
rcl_node_init: multiple_executors_2 0x7fded41a9970
rmw_create_node: multiple_executors_2
rcl_node_fini: 0x7fded41a9970
rmw_destroy_node: 0x7fded41a74a0
return OK
rcl_node_fini: 0x7fded3d59ae0
rmw_destroy_node: 0x7fded417bba0
return OK
[       OK ] test_executor__rmw_opensplice_cpp.multiple_executors (81 ms)
[ RUN      ] test_executor__rmw_opensplice_cpp.notify
rcl_node_init: test_executor_notify 0x7fded447f160
rmw_create_node: test_executor_notify
Waited 300 microseconds for the subscriber to connect to topic 'test_executor_notify_subscription'
rcl_node_fini: 0x7fded447f160
rmw_destroy_node: 0x7fded43bf5e0
return OK
[       OK ] test_executor__rmw_opensplice_cpp.notify (44 ms)
[----------] 4 tests from test_executor__rmw_opensplice_cpp (179 ms total)

[----------] Global test environment tear-down
[==========] 4 tests from 1 test case ran. (180 ms total)
[  PASSED  ] 4 tests.
```

gerkey, 2016-06-08:
Perhaps relevant: I have backtrace from a repeated run that crashed, in which two threads are simultaneously at this point:

```
frame #5: 0x000000010486b57d librmw_opensplice_cpp.dylib`::rmw_wait(subscriptions=0x00007fbd91801808, guard_conditions=0x00007fbd91801820, services=0x00007fbd91801850, clients=0x00007fbd91801838, waitset=0x00007fbd93b0e8e0, wait_timeout=0x00000001079c1d60) + 3613 at rmw_wait.cpp:298
   295 	          return RMW_RET_ERROR;
   296 	        }
   297 	      }
-> 298 	      if (dds_waitset->detach_condition(guard_condition) != DDS::RETCODE_OK) {
   299 	        RMW_SET_ERROR_MSG("failed to detach guard condition");
   300 	        return RMW_RET_ERROR;
   301 	      }
```

The two threads are operating on different `dds_waitset` pointers, but they're operating on the same `guard_condition` pointer. I.e., two threads are trying simultaneously detach the same guard condition from two different wait sets.

wjwwood, 2016-06-08:
Hmm that shouldn't be allowed. Wait sets should have different contents always. It would be good to back that out and figure out why they share a guard condition. Maybe two executors at the same time?

gerkey, 2016-06-08:
Yep, definitely multiple executors.

Thread 1:
```
frame #13: 0x00000001032c10a2 gtest_executor__rmw_opensplice_cpp`test_executor__rmw_opensplice_cpp_multiple_executors_Test::TestBody(this=0x00007fbd9386d8b0) + 1602 at test_executor.cpp:121
   118 	  // Launch both executors
   119 	  std::thread execution_thread(spin_executor2);
   120 	
-> 121 	  executor1.spin();
   122 	  execution_thread.join();
   123 	  EXPECT_EQ(counter1.load(), counter_goal);
   124 	  EXPECT_EQ(counter2.load(), counter_goal);
```
Thread 2:
```
frame #10: 0x00000001032cd2a8 gtest_executor__rmw_opensplice_cpp`test_executor__rmw_opensplice_cpp_multiple_executors_Test::TestBody(this=0x00007fbd93ac6150)::$_5::operator()() const + 24 at test_executor.cpp:115
   112 	  executor2.add_node(node2);
   113 	
   114 	  auto spin_executor2 = [&executor2]() {
-> 115 	      executor2.spin();
   116 	    };
   117 	
   118 	  // Launch both executors
```

wjwwood, 2016-06-08:
Ok, then it must be the signal guard condition. We'll have to create a guard condition for each executor and trigger them all on sigint or shutdown.

gerkey, 2016-06-08:
Where is the signal guard condition created? I can try modifying that code to create one per wait set.

gerkey, 2016-06-08:
Are we talking about this static?

https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L38-L40

jacquelinekay, 2016-06-08:
Yep.

You could make a global singleton vector of signal guard conditions. Or you could associate it with the "Context" object.

I think it makes sense to have one guard condition per executor, but we should also ensure that the access to each guard condition is single-threaded (since an executor could have multiple threads).

gerkey, 2016-06-08:
Having sufficiently peeled the onion, I can confirm that the guard condition in question is the global static signal guard condition:

```
(lldb) p *g_sigint_guard_cond_handle.impl->rmw_handle
(rmw_guard_condition_t) $21 = (implementation_identifier = "rmw_opensplice_cpp", data = 0x00007fbd91405c40)
(lldb) p guard_condition
(DDS::GuardCondition *) $22 = 0x00007fbd91405c40
```

tfoote, 2016-06-08:
On the thread leakage: 

I was working my way up the stack from the bottom. I put a breakpoint on `u_particpantDeinit` (it is what cleans up the threads) and it never hit except in error cases.

We are calling destroy but this line skips the destroy because the participant has a reference count of 2: https://github.com/osrf/opensplice/blob/master/src/user/code/u_participant.c#L693 Which is why the threads are not getting cleaned up. Forcing that if statement cleans up the threads but we need to find out why there's an extra reference count.

wjwwood, 2016-06-08:
@gerkey ok, well the rule "wait sets should not contain overlapping elements" was only recently clarified: https://github.com/ros2/rcl/pull/59

Currently, when you create an executor it creates a wait set and always adds the global signal guard condition, as you said. What I think we need to transition that to is: when you create an executor it creates a guard condition and registers it with the signal handling code and the signal handler triggers all registered guard conditions on a sigint or shutdown.

This is a similar pattern that I've had to adopt with the graph listening stuff since the lone graph guard condition that each node has can only be waited on by one wait set.

There's already a mutex for the shutdown state which is used by the condition variable that the `rclcpp::sleep_for` function uses. It would be fine to reuse that mutex to lock a global vector rcl guard conditions which would replace the single global guard condition. Then add register and unregister functions when would acquire the mutex and add or remove the guard conditions to the list to be triggered on sigint or shutdown.

@gerkey are you going to try and address this now or just ticket it? I'd be happy to look into it as soon as I have time. Thanks for tracking it down.

gerkey, 2016-06-08:
Quick and dirty implementation of giving a unique signal guard condition to each waitset:

https://gist.github.com/gerkey/d5b8f9f453f3324effcd86d76167ec35

With that patch applied locally, after limited testing, I'm no longer getting crashes (which I was getting reliably). Now I'm getting error messages that point to the thread leakage that Tully looked into:

```
========================================================================================
Report      : WARNING
Date        : Wed Jun 08 13:30:17 PDT 2016
Description : pthread_create failed with error 35 (gtest_executor__rmw_opensplice_cpp <28285>)
Node        : Brians-MacBook-Air.local
Process     : gtest_executor__rmw_opensplice_cpp <28285>
Thread      : main thread 7fff7b29a300
Internals   : V6.4.140407OSS///os_threadCreate/os_thread.c/481/2/1465417817.373031000
========================================================================================
```

gerkey, 2016-06-08:
@wjwwood Thanks for the pointer to the existing pattern. For now I'm doing the minimum necessary to determine where the problems are. Once I have everything worked out, I'll come back and do it properly.

wjwwood, 2016-06-08:
👍

gerkey, 2016-06-08:
Some progress to report. My test setup is:

* `rclcpp` on branch `opensplice_cleanup` (to give each waitset its own signal guard condition, which fixes a crash that's unrelated to the thread leakage, but is required to be able to run repeated tests)
* `system_tests` on branch `opensplice_cleanup` (to comment out some test cases to allow repeating just the simplest one)
* `rmw_opensplice` on branch `opensplice_cleanup` (to comment out the use of the builtin subscriber during node creation)

With all of those changes, I can execute the simplified test repeatedly (`../../../../build/test_rclcpp/gtest_executor__rmw_opensplice_cpp --gtest_repeat=-1`) with no increase in per-test execution time (constant at ~3ms) and no other apparent increase in system resources. That's the first time that I've been able to repeatedly run an opensplice-based test without growth in resource use.

From this result, I am inclined to believe that the problem is with our management of the `builtin_subscriber` and related entities. We work with them here during node creation:

https://github.com/ros2/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_node.cpp#L131-L179

I'm not sure, but I think that we're not correctly doing the symmetric operation during node destruction:

https://github.com/ros2/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_node.cpp#L281-L311

and that, as a result, the underlying participant is never actually being deleted.

If @wjwwood or somebody else who's familiar with this creation and destruction logic wants to take a look, please do. I'm done for today; I'll try to come back to it tomorrow afternoon.

tfoote, 2016-06-08:
I can confirm that commenting out the builtin_subscriber/publisher content gets rid of the thread leakage. With those sections enabled the refcount on the publisher is 2 at the end of `rmw_create_node` The solution is to explicitly delete the builtin subscriber. 

I'll update the branch to add that instead of commenting out the sections.

gerkey, 2016-06-13:
#144 improves cleanup in OpenSplice. It doesn't specifically address the "Handle Server" error that this is issue was originally about, but in all of my testing over the past week, I haven't encountered that error. So I'm going to close this issue, on the assumption that it was somehow fixed along the way. Re-open if you see it again.

closed 2016-06-13 by gerkey