ros2/rmw_opensplice #124: DDS Configuration - Localhost

firesurfer, 2016-03-28:
I tried to configure opensplice to use localhost as interface if possible. If I put localhost in the ospl.xml as network interface it states that I have to use an interface that supports multicast. 

I think this might relate to: #115 
Is it possible to configure opensplice somehow, that it uses localhost for communication between processes on the same host? My search via google didn't show many results for this issue.

wjwwood, 2016-03-28:
@firesurfer I haven't had time to look into this issue or #115. Can you give some details about how you investigated this? Did you use wireshark or the log output of OpenSplice to determine whether or not the loopback is being used? (I assume you mean the loopback interface when you say localhost)

firesurfer, 2016-03-29:
How I figured out which interface opensplice uses:
It tells you at the start of the program:
```
Report      : WARNING
Date        : Tue Mar 29 10:51:11 CEST 2016
Description : using network interface wlan0 (192.168.208.116) selected arbitrarily from: wlan0, vmnet1, vmnet8
```

How I figured out that all network traffic goes over this interface (even traffic that has its endpoint on the same machine:
If I use a cable based network connection instead of wireless lan I don't have any transport delays or data loss.

I did some more research after submitting this issue and found this thread in the opensplice forum:
http://forums.opensplice.org/index.php?/topic/2604-no-message-delivered-when-there-is-no-network-connection/

Apparently it is necessary to enable multicast on loopback, in order to have opensplice use this interface. But in this configuration opensplice (community edition) won't talk to other computers running dds. (At least I think so, based on the forum thread) It seams that many useful features are reserved for the commercial version of opensplice, but perhaps I didn't find the correct configuration.

wjwwood, 2016-03-30:
Yeah, a somewhat related issue was discussed between RTI and Tinq developers after we brought up the issue, see: https://github.com/ros2/ros2_embedded_nuttx/issues/26#issuecomment-64503031 It's an evolving issue.

> It seams [sic] that many useful features are reserved for the commercial version of opensplice, but perhaps I didn't find the correct configuration.

There are a few features (like "real-time" support) that they don't offer in their community edition. We're working with them and other DDS vendors to see what we can make available under their open licenses, which will impact which one we decide to make the "default" version we send with our binaries. I think it is in their interest to provide any features that affect performance in their open edition, so hopefully we can convince them of that. It's also the case that the shared memory option provided by RTI doesn't work with other DDS vendors, for example. We're keeping an eye on these drawbacks and seeing what we can do about each of them as we go.

dhood, 2018-02-22:
we'll close this under the assumption that opensplice 6.7 addresses the issue but can reopen it if that's not the case

closed 2018-02-22 by dhood