ros2/rmw_opensplice #128: memory leaks: are they fixable?

jacquelinekay, 2016-04-20:
According to valgrind, rclcpp programs using rmw_opensplice_cpp leak memory.

An example output from gtest_publisher__rmw_opensplice_cpp in test_rclcpp leaks 5kb for sure and possibly up to 37kb of memory:

```
$ valgrind --leak-check=full -v build/test_rclcpp/gtest_publisher__rmw_opensplice_cpp
==4894== LEAK SUMMARY:
==4894==    definitely lost: 5,215 bytes in 95 blocks
==4894==    indirectly lost: 32,923 bytes in 1,221 blocks
==4894==      possibly lost: 1,036,095 bytes in 15,819 blocks
==4894==    still reachable: 1,582,502 bytes in 1,643 blocks
==4894==         suppressed: 0 bytes in 0 blocks
```

Compare this to the same executable built with Connext, which reports no bytes definitely or indirectly lost (after ros2/rmw_connext#155 was merged):
```
==5952== LEAK SUMMARY:
==5952==    definitely lost: 0 bytes in 0 blocks
==5952==    indirectly lost: 0 bytes in 0 blocks
==5952==      possibly lost: 185,663 bytes in 2,192 blocks
==5952==    still reachable: 649,472 bytes in 2,899 blocks
==5952==         suppressed: 0 bytes in 0 blocks
```

The optimistic explanation is that the Opensplice garbage collector is actually taking care of the leaked memory for us. It would be nice to confirm what's actually happening and if we need to do a better job cleaning up memory somewhere in Opensplice.

(It's also interesting to compare the memory used in Connext vs. Opensplice.)

dirk-thomas, 2016-04-20:
You might want to run valgrind but with options to additionally log where this memory has been allocated. With that information it is usually much easier to tell if / how to resolve it.

gerkey, 2016-06-13:
Duplicate of #145.

closed 2016-06-13 by gerkey