ros2/rmw_opensplice #149: Crash when used in dynamic_bridge

gerkey, 2016-06-28:
I'm seeing this crash pretty often when bridging from ROS 2 to ROS 1 and the ROS 1 listener exits:

```
removed 2to1 bridge for topic 'image'
terminate called after throwing an instance of 'std::runtime_error'
  what():  rcl_wait() failed: read condition handle is null, at /home/gerkey/ros2_ws_debug/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_wait.cpp:266, at /home/gerkey/ros2_ws_debug/src/ros2/rcl/rcl/src/rcl/wait.c:581
```

Backtrace:
```
(gdb) bt
#0  0x00007ffff4b08418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff4b0a01a in __GI_abort () at abort.c:89
#2  0x00007ffff514184d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff513f6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff513f701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff513f919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff74869e3 in rclcpp::executor::Executor::wait_for_work (this=0x7fffffffd890, timeout=...) at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:394
#7  0x00007ffff7487b15 in rclcpp::executor::Executor::get_next_executable (this=0x7fffffffd890, timeout=...) at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:525
#8  0x00007ffff74856de in rclcpp::executor::Executor::spin_once (this=0x7fffffffd890, timeout=...) at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:180
#9  0x00007ffff7485204 in rclcpp::executor::Executor::spin_node_once_nanoseconds (this=0x7fffffffd890, node=std::shared_ptr (count 8, weak 2) 0x666090, timeout=...)
    at /home/gerkey/ros2_ws_debug/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:148
#10 0x000000000042ff7e in rclcpp::executor::Executor::spin_node_once<std::ratio<1l, 1000l> > (this=0x7fffffffd890, node=std::shared_ptr (count 8, weak 2) 0x666090, timeout=...)
    at /home/gerkey/ros2_ws_debug/install/include/rclcpp/executor.hpp:145
#11 0x000000000042b632 in main (argc=1, argv=0x7fffffffdb18) at /home/gerkey/ros2_ws_debug/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:427
```

BrannonKing, 2017-04-05:
So does dynamic_bridge run indefinitely with fast-rtps? I had assumed that the bridge's crashing every few minutes was a bridge bug, not an rmw_opensplice bug.

BrannonKing, 2017-04-05:
In looking at the code (below), it seems we are contrary to the documentation in the header (types.h). The header says that entries in the `subscriptions->subscribers` can come and go; the entry in the array is expected to be null at times. Hence, the first error should be a `continue` instead (and this code is repeated elsewhere in the file). 

Concerning the second error, when the read condition is null: I can't see how that can happen in the subscription creator. I didn't track down where subscriptions are added and removed, but I hope they are constructed before they are added to the subscribers list and that they are destructed after the entry in the subscribers list has been nulled. That could still leave us in a situation where the reader is being destructed after we've taken a pointer on it. We could address this through some concurrency methods, either locking or immutable structures. Alternately, we could just assume that a null `read_condition` means that this subscriber is in process of teardown and call `continue` instead of throwing an error.

```cpp
  for (size_t i = 0; i < subscriptions->subscriber_count; ++i) {
    OpenSpliceStaticSubscriberInfo * subscriber_info =
      static_cast<OpenSpliceStaticSubscriberInfo *>(subscriptions->subscribers[i]);
    if (!subscriber_info) {
      RMW_SET_ERROR_MSG("subscriber info handle is null");
      return RMW_RET_ERROR;
    }
    DDS::ReadCondition * read_condition = subscriber_info->read_condition;
    if (!read_condition) {
      RMW_SET_ERROR_MSG("read condition handle is null");
      return RMW_RET_ERROR;
    }
```

dirk-thomas, 2017-09-25:
@gerkey @BrannonKing OpenSplice wasn't support for quite a while in ROS 2. Since the recent Beta 3 it is supported again (at least on Linux and Windows without security). Can either of you confirm that the problem still exists in the current release.If yes, please also include some instructions how to reproduce this. Otherwise we might close this ticket for now.

gerkey, 2017-09-26:
Let's close it and reopen if someone can demonstrate the problem again.

closed 2017-09-26 by gerkey