ros2/rmw_opensplice #156: Bad warning when opensplice is not found

geoffviola, 2016-11-07:
If I don't install or use opensplice on Windows, the only indication that something might be wrong is a "The system cannot find the path specified." message. This message occurs when C:\dev\ros2\setup.bat is called. Ament works fine by itself at this point. Since the two packages are distributed independently, there should be a better error message. But should the .bat file exit with a bad error code?

This issue is related to the line at https://github.com/ros2/rmw_opensplice/blob/919206678d6028edb56d5c99aa99434bd58edba4/opensplice_cmake_module/env_hook/opensplice.bat.in#L3.

wjwwood, 2017-02-09:
This env hook should not be called unless OpenSplice is found with `find_package()`, based on the CMake code:

https://github.com/ros2/rmw_opensplice/blob/919206678d6028edb56d5c99aa99434bd58edba4/opensplice_cmake_module/CMakeLists.txt#L16

I'm not sure why it's getting called in the first place. I'd say the right thing to do is to ensure this env hook isn't called by the `setup.bat` in the first place rather than trying to add a different error message.

dhood, 2018-02-22:
sounds like the question here is if it was reasonable to call the `setup.bat` or not. https://github.com/ros2/ros2/issues/399 improved support for "optional" rmw implementations, and the December 2017 Ardent release included opensplice+fastrtps fat binaries for windows that optionally supported opensplice installations. In the testing of those, we didn't come across this issue again so I will close this for now. If it comes up again and has an identified cause we can reopen

closed 2018-02-22 by dhood