ros2/rmw_opensplice #165: rcl_init called multiple times

BrannonKing, 2017-03-30:
I'm getting the error below when attempting to use rmw_opensplice in the current (Mar 29) codebase. I'm running a Gazebo simulation. I'm guessing that Gazebo loads all the plugins into the same process, which causes this error. It seems I can eat the error and still run fine. I just wondered if we should track the initialization state in the RMW.

```
[rmw|error_handling.c:93] error string being overwritten: rcl_init called while already initialized, at src/ros2/rcl/rcl/src/rcl/rcl.c:69
failed to initialize rmw implementation: rcl_init called while already initialized, at src/ros2/rcl/rcl/src/rcl/rcl.c:69
```

BrannonKing, 2017-08-31:
I suppose catching the exception in Gazebo is fine. I just wanted the logging removed (since it already throws an exception). BTW, this should move to rcl; it's logged in the wrong spot.

dirk-thomas, 2017-08-31:
This issue was moved to ros2/rcl#162

closed 2017-08-31 by dirk-thomas