ros2/rmw_opensplice #171: recompliation of messages always happens

BrannonKing, 2017-04-20:
Whenever my message projects are recompiled (using Ament), regardless of whether any code has changed or not, the rmw_opensplice message support gets rebuilt. This is painful and expensive, costing me an hour of compilation time every day. Is there something I can do to make middleware support be smart enough to not recompile message packages when nothing has changed?

mikaelarguedas, 2017-04-20:
Yes we are aware of this issue indeed. This (re)generation + recompilation of messages started to happen since https://github.com/ros2/rmw_opensplice/pull/158 and https://github.com/ros2/rmw_opensplice/pull/161 have been merged.
Given that we don't actively support opensplice we haven't take the time to investigate it.

I think the regeneration happens because several generated files (not generated by the cmake custom command) are listed as such. a good place to start digging would be https://github.com/ros2/rmw_opensplice/blob/0b18b700ba44834b81916e007e67ab0457d47c4e/rosidl_typesupport_opensplice_c/cmake/rosidl_typesupport_opensplice_c_generate_interfaces.cmake#L199 and https://github.com/ros2/rmw_opensplice/blob/0b18b700ba44834b81916e007e67ab0457d47c4e/rosidl_typesupport_opensplice_c/cmake/rosidl_typesupport_opensplice_c_generate_interfaces.cmake#L221. 
This list `_generated_msg_files` is the one used to trigger the custom_command https://github.com/ros2/rmw_opensplice/blob/0b18b700ba44834b81916e007e67ab0457d47c4e/rosidl_typesupport_opensplice_c/cmake/rosidl_typesupport_opensplice_c_generate_interfaces.cmake#L175 but I don't think that the custom_command actually generate them.

We're happy to take PRs to address this issue

BrannonKing, 2017-04-20:
So do the other middlewares suffer from the same issue?

mikaelarguedas, 2017-04-20:
no, the opensplice message generation is the only one suffering from it, you won't experience this if you're using RTI Connext or Fast-RTPS

BrannonKing, 2017-04-20:
It might be okay if the custom command runs every time; it's probably a lot faster than the compiler. Do we just need to make the file generator smart enough to not overwrite files if the newly generated one is identical?

wjwwood, 2017-04-21:
@BrannonKing Yeah, it's something like that. Probably a custom command in cmake that is not setup with the correct dependencies to files. For example, if they depended on a file that never gets created then it would be run every time.

mikaelarguedas, 2017-04-21:
> @BrannonKing Yeah, it's something like that. Probably a custom command in cmake that is not setup with the correct dependencies to files. For example, if they depended on a file that never gets created then it would be run every time.

:+1: I think `_generated_msg_files` should not have any of the files listed in the pointers I gave in https://github.com/ros2/rmw_opensplice/issues/171#issuecomment-295948276
This may not be the only thing required to fix this but I think is a good starting point

PR #192 (2017-08-31): fix wrong expectation about output srv files
Fixes #171.
closed 2017-08-31 by dirk-thomas