ros2/rmw_opensplice #177: Common topic name split function required

michielb-prismtech, 2017-07-20:
I'm trying to fix issue in the rmw_opensplice implementation based on these test results http://ci.ros2.org/job/ci_linux/2882/#showFailuresLink

I'm running in to an issue with the way the topic_name is defined in ros2. This does not match the OMG specified DDS APIs. ros2 define a topic as a partition_name/topic_name so for all spec compliant DDS APIs a function is required to split up the names. With a previous PR I've added a function to rmw_opensplice_cpp which does just that based on a function in the rmw_connext repository, however I now need it in the rosidl_typesupport_opensplice_cpp as well can don't have access to the previously added function. For test purposed I've locally added a rcutils_split_topic function for spliting to the rcutils to I'm unsure where to put this common split function. Can you help me with that?

dirk-thomas, 2017-07-24:
@wjwwood Maybe you can give an advice here since you implemented the support for FastRTPS as well as Connext.

wjwwood, 2017-07-24:
Actually @Karsten1987 implemented this part, but I think I can answer it. 

Well having a common function to split the topic name into parts would be fine, but it definitely should not go into rcutils, since that package needs to stay ros and dds agnostic. 

We might need a `rmw_dds` package, sort of like we already have `rosidl_dds`. I don't mind either way, but maybe we could put it in the `rosidl_dds` to avoid making another package. Unless there is a reason `rmw_opensplice_cpp` cannot depend on it.

wjwwood, 2017-07-24:
Sorry, we have a `rosidl_dds` repo, but not a package called that. Maybe it could be squeezed into `rosidl_generator_dds_idl`, but that seems like abusing that package. As I said I don't mind it to avoid another package but creating a new package would be fine too. 

Also the code is quite trivial so copying, while not ideal, is probably ok too.

michielb-prismtech, 2017-07-26:
I've solved this issue by just fixing it in the rmw_opensplice repository, so other rmw's won't benefit nor suffer from it. #179

closed 2017-07-26 by michielb-prismtech