ros2/rmw_opensplice #199: implement ``

dirk-thomas, 2017-09-11:
Currently the rmw function `rmw_service_names_and_types` is not implemented (see https://github.com/ros2/rmw_opensplice/blob/ee19bdf5957ff1ac7384604583d43feacae2e6b5/rmw_opensplice_cpp/src/rmw_service_names_and_types.cpp#L34).

This is necessary to support the `ros1_bridge`.

mikaelarguedas, 2017-09-11:
Note: A test for this could be added to the ros2cli test suite to catch these errors in regular ci and not only on the packaging job

michielb-prismtech, 2017-09-12:
Working on the implementation, can takers for the testcase?

dirk-thomas, 2017-09-12:
> can takers for the testcase?

Can you please clarify the question.

You can call `ros2 service list` on the command line to try the API. Maybe run a server / client at the same time.

michielb-prismtech, 2017-09-12:
>Can you please clarify the question.
more of a response to 
>Note: A test for this could be added to the ros2cli test suite to catch these errors in regular ci and not only on the packaging job

dirk-thomas, 2017-09-12:
@michielb-prismtech You don't have to worry about the test coverage for this. If it works for you with the command line tool that would be great.

wjwwood, 2017-09-12:
For anyone who is interested, the tests for this function's "sister" function are here:

https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_graph.cpp#L122-L158

Probably could be duplicated and slightly modified for coverage of this function.

dirk-thomas, 2017-09-12:
We might have to release Beta 3 without this change - depending on the time frame. We plan to announce the release tomorrow.

dirk-thomas, 2017-09-14:
Addressed by #200.

closed 2017-09-14 by dirk-thomas