ros2/rmw_opensplice #202: OSX Symbol not found: __cg_jpeg_resync_to_restart

sloretz, 2017-09-13:
Not sure if this belongs here or on ros2/build_cop

## Bug report

**Required Info:**

- Operating System:
  - OSX (dosa)
    - ProductName:	Mac OS X
    - ProductVersion:	10.12.6
    - BuildVersion:	16G29

- Installation type:
  - packaging_osx job
- Version or commit hash:
  - ee19bdf5957ff1ac7384604583d43feacae2e6b5
- DDS implementation:
  - Opensplice
- Client library (if applicable):
  - NA

#### Steps to reproduce issue

http://ci.ros2.org/view/packaging/job/packaging_osx/740/
Run a packaging_osx job with `use_opensplice: true` or ssh into a machine that ran the failed job.

```
$ pwd
/Users/osrf/jenkins/workspace/packaging_osx/ws/build/rosidl_typesupport_opensplice_cpp
$ . /Users/osrf/jenkins/workspace/packaging_osx/ws/build/rosidl_typesupport_opensplice_cpp/cmake__build.sh && /usr/local/bin/cmake /Users/osrf/jenkins/workspace/packaging_osx/ws/src/ros2/rmw_opensplice/rosidl_typesupport_opensplice_cpp -DBUILD_TESTING=0 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DSECURITY=ON -DCMAKE_INSTALL_PREFIX=/Users/osrf/jenkins/workspace/packaging_osx/ws/install -G Unix Makefiles
```

#### Expected behavior
Build would succeed

#### Actual behavior

```
dyld: Symbol not found: __cg_jpeg_resync_to_restart
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO
  Expected in: /usr/local/lib/libJPEG.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO
Abort trap: 6
```

#### Additional information

This is odd. It uses a system library `ImageIO` which was built against a system version of libjpeg. libjpeg was also installed via brew (why?). They have different symbols, but the brew installed one is the one found first.

libjpeg.dylib comes from brew

```$ ls -lh /usr/local/lib/libjpeg.dylib 
lrwxr-xr-x  1 osrf  admin    35B Sep  6 20:52 /usr/local/lib/libjpeg.dylib -> ../Cellar/jpeg/9b/lib/libjpeg.dylib
```

It was updated recently (september 6th)

```
$ brew info libjpeg
jpeg: stable 9b (bottled)
Image manipulation library
http://www.ijg.org
/usr/local/Cellar/jpeg/8d (19 files, 730.9KB)
  Poured from bottle on 2014-10-17 at 13:30:18
/usr/local/Cellar/jpeg/9b (20 files, 724KB) *
  Poured from bottle on 2017-09-06 at 20:52:53
From: /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/jpeg.rb
```

It has these symbols
[... 292 lines cut ...]
0000000000018124 T __cg_jpeg_write_tables
0000000000008081 T __cg_jround_up
                 U _asprintf
                 U _exit
                 U _fclose
                 U _ferror
                 U _fflush
                 U _fopen
                 U _fprintf
                 U _fread
                 U _free
                 U _fseek
                 U _fwrite
                 U _malloc
                 U _memcpy
                 U _memset
                 U _memset_pattern16
                 U _mktemp
                 U _posix_memalign
                 U _strncmp
                 U _strncpy
                 U _tmpfile
                 U _unlink
                 U dyld_stub_binder
```

dirk-thomas, 2017-09-13:
We currently don't target Mac OS with OpenSplice 6.7 - only Linux and Windows. I assume the OS X node has still OpenSplice 6.4 installed.

sloretz, 2017-09-13:
Yes it's OpenSplice 6.4. Does that mean this is a problem that can be ignored for the time being?

```
$ brew info opensplice
osrf/ros2/opensplice: stable 6.4.0, HEAD
http://www.prismtech.com/opensplice
/usr/local/Cellar/opensplice/6.4.0 (795 files, 10.2MB) *
  Built from source on 2016-01-24 at 11:15:59
From: /usr/local/Homebrew/Library/Taps/osrf/homebrew-ros2/opensplice.rb
==> Dependencies
Build: cmake ✔
Required: gawk ✔
==> Requirements
Build: java ✘
==> Options
--with-debug
	Builds opensplice in debug mode
--HEAD
	Install HEAD version
```

mikaelarguedas, 2017-09-13:
> Does that mean this is a problem that can be ignored for the time being?

yes

closed 2017-09-13 by dirk-thomas