ros2/rmw_opensplice #205: Opensplice malformed packet warning

kengregson, 2017-09-22:
Using Ubuntu16.04 with binary (apt/dpkg) install of R2B3 and OSRF built opensplice packages seems to have some interoperability issues with other DDS implementations

When running the ros1 bridge example with Opensplice, if I start up a (ROS2) publisher which uses the built in eProsima FastRTPS  on the /chatter topic via:
ros2 topic pub /chatter std_msgs/String "{data: Hi}"'

The dynamic bridge and listener complain:(twice)
========================================================================================
Report      : WARNING
Date        : Fri Sep 22 10:21:24 EDT 2017
Description : malformed packet received from vendor 1.15 state parse:heartbeat <52545053 0201010f 76050f01 00005536 00000000 0e010c00 3dd442e5 0000023d @0x24 07011c00 04050000 03010000 00000000 02000000 00000000 00000000 01000000> (note: maybe partially bswap'd) {{7,1,28},504,103,2,0}

Node        : ultrazen
Process     : dynamic_bridge <14900>
Thread      : recv 7f404ae98700
Internals   : 6.7.170523OSS///ddsi2/q_receive.c/2939/0/1506090084.832221836/0

But things still seem to work.

Opensplice always complains on startup that:
========================================================================================
Report      : ERROR
Date        : Fri Sep 22 10:12:45 EDT 2017
Description : Failed to open configuration syntax file: /usr/etc/metaconfig.xml
Node        : ultrazen
Process     : dynamic_bridge <14900>
Thread      : spliced 7f405d20f700
Internals   : 6.7.170523OSS///configuration validator/cfg_validator.c/825/0/1506089565.576414694/0

This can be reliably reproduced, I set up these bash aliases to save typing:
# setup ROS1 or/and ROS2
alias r1='source /opt/ros/kinetic/setup.bash'
alias r2='source /opt/ros/r2b3/setup.bash;source /opt/ros/r2b3/share/ros2cli/environment/ros2-argcomplete.bash'
# optionally use opensplice instead of FastRTPS
alias os='export RMW_IMPLEMENTATION=rmw_opensplice_cpp'
# ROS1-2 bridge example nodes
alias n1='r1;roscore'
alias n2='r1;r2;ros2 run ros1_bridge dynamic_bridge'
alias n3='r2;ros2 run demo_nodes_cpp listener'
alias n4='r1;rosrun rospy_tutorials talker'
# second ROS2 /chatter publisher
alias n5='r2;ros2 topic pub /chatter std_msgs/String "{data: Hi}"'

- Step 1, terminal 1: start roscore: ">n1"
- Step 2, terminal 2: start ros1-2 bridge using opensplice: ">os;n2"
- Step 3, terminal 3: start ros2 listener (cpp) using opensplice: ">os;n3"
- Step 4, terminal 4: start ros1 talker (python) using opensplice: ">os;n4"

Things start running as expected with matching "Publishing"/"I heard" output.

- Step 5, terminal 5: start second /chatter publisher (ros2) using FastRTPS: ">n5"
Things appear to work in spite of the warning.

Same result can be obtained with only a single /chatter publisher (and no step 5) by:
- Step 4, terminal 4: start ROS2 talker (cpp) under FastRTPS: ">ros2 run demo_nodes_cpp talker"

The warning does not seem to occur when all ROS2 nodes are run exclusively under either FastRTPS or Opensplice or when running the bridge and listener with FastRTPS and talker with Opensplice

michielb-prismtech, 2017-09-26:
I've seen:
> Opensplice always complains on startup that:

Which means the metaconfig.xml is missing and as a result the configuration cannot be validated before the initialization phase. Since the default configuration is checked this is not a problem, but when making incorrect changes it could lead to undesired behavior. Solution would be to add the metaconfig.xml to the OpenSplice dpkg. (In OpenSplice 6.8 and higher the name of the file is changed to ospl_metaconfig.xml)

eboasson, 2018-01-05:
The message

> Description : malformed packet received from vendor 1.15 state parse:heartbeat <52545053 0201010f 76050f01 00005536 00000000 0e010c00 3dd442e5 0000023d @0x24 07011c00 04050000 03010000 00000000 02000000 00000000 00000000 01000000> (note: maybe partially bswap'd) {{7,1,28},504,103,2,0}

is because a DDSI message received contains an invalid "heartbeat" message. There are two things wrong with the message: the last two numbers are the "first sequence number" and the "last sequence number". According to the spec, 1 <= first <= last. The spec furthermore requires dropping it when it doesn't meet this requirement ...

I have been adding workarounds over the years for known cases (at least when OpenSplice is in its default "lax" mode), but I wasn't aware that fastRTPS sent them like this ... The case I know of is that the first sequence number is >= 1 and the last is >= first-1. This is particular form is going to be allowed in an update of the spec, but it still won't cover the (2,0) received here.

So I think it should be filed as a bug against fastRTPS.

dirk-thomas, 2018-01-05:
Can either of you please fill a ticket against [FastRTPS](https://github.com/eProsima/Fast-RTPS) and reference it here.

dhood, 2018-02-22:
waiting for response from fastrtps in https://github.com/eProsima/Fast-RTPS/issues/177

KevKzTW, 2018-03-07:
Hi ,

sudo ln -s /usr/etc/opensplice/metaconfig.l /usr/etc
should fix problem 
> Opensplice always complains on startup that:

dirk-thomas, 2018-03-08:
Placing the file into `/usr/etc/metaconfig.xml` seems weird to me. That could easily collide with other packages.

@michielb-prismtech Can you comment on that suggestion? Is that where we should place the `metaconfig.xml` in the Debian package in the first place? Or is there a way to make it use the file in the location it currently is?

michielb-prismtech, 2018-03-15:
We agree that the name file name metaconfig.xml is too generic, that's why in 6.8 we've changed it to ospl_metaconfig.xml. OpenSplice looks for the metaconfig.xml file in $OSPL_HOME/etc with our installer this is a location which does not collide. 
@dirk-thomas Would the rename to ospl_metaconfig.xml solve your problem?

dirk-thomas, 2018-03-16:
> Would the rename to ospl_metaconfig.xml solve your problem?

@michielb-prismtech If that will be done for 6.8 that should be fine. Until then we have to generate a new Debian package of OpenSplice 6.7 which we use that moves the location of the file (even though the name is not optimal).

dirk-thomas, 2018-03-16:
@kengregson Can you please try the latest version of FastRTPS which is supposed to fix the problem and report back about your findings. Thanks.

dirk-thomas, 2018-03-20:
The warning about the config file has been addressed in the latest Debian package which adds the symlink as proposed above: `libopensplice67_6.7.0+osrf2-3`.

dirk-thomas, 2018-03-23:
I will close this ticket now since it looks like we addressed the problems. @kengregson please follow up on the referenced FastRTPS issue [177](https://github.com/eProsima/Fast-RTPS/issues/177) to confirm it addresses your problem.

closed 2018-03-23 by dirk-thomas