ros2/rmw_opensplice #242: ros2 node list could not ignore the once exited node (participant)

reed-lau, 2018-08-21:
## Bug report
- Operating System:
  -Ubuntu 16.04
- Installation type:
   from source  
- Version or commit hash:
  - ca16eedb396cb069dac7501469a0d0a6a146a4a1  0.5.2 bouncy
- DDS implementation:
  - opensplice
- Client library (if applicable):
  - rmw_opensplice or rclpy

#### Steps to reproduce issue
1.    ros2 node list ( start the daemon)
2.    run talker ( from demo_nodes_cpp), then Ctrl-C it
3.   ros2 node list
the talker is still there

#### Expected behavior
when there's no node running, ros2 node list should list nothing.

#### Actual behavior
the once exited node ever since the daemon has run in background.

#### Additional information
the Opensplice's api get_discovered_participant/get_discovered_participant_data could get all the information ever since it join into DDS domain.

https://github.com/ros2/rmw_opensplice/blob/ca16eedb396cb069dac7501469a0d0a6a146a4a1/rmw_opensplice_cpp/src/rmw_node_names.cpp#L71

https://github.com/ros2/rmw_opensplice/blob/ca16eedb396cb069dac7501469a0d0a6a146a4a1/rmw_opensplice_cpp/src/rmw_node_names.cpp#L88

mikaelarguedas, 2018-08-21:
Thanks @reed-lau for reporting.

This is currently tracked at https://github.com/ros2/ros2/issues/514 as the problem spans more than one DDS implementation.

Marking this issue as duplicate

cwyark, 2019-06-20:
I got the same problem when using dashing from deb. 
Operating System:
-Ubuntu 18.04
Installation type:
from deb 
Version or commit hash:
-
DDS implementation:
opensplice
Client library (if applicable):
rmw_opensplice / rclcpp

Steps to reproduce issue
ros2 node list ( start the daemon)
run talker ( from demo_nodes_cpp), then Ctrl-C it
ros2 node list
the talker is still there

dirk-thomas, 2019-08-01:
Assuming this was addressed by #275 I am going to close this.

closed 2019-08-01 by dirk-thomas