ros2/rmw_opensplice #254: Bad parameter: Could not register type 'lifecycle_msgs::srv::dds_::Sample_ChangeState_Request_'

ahcorde, 2018-12-20:
Hi, I have compiled ros2 crystal from sources in Ubuntu 18.04, I'm not able to execute lifecycle examples.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from sources, native build
- Version or commit hash:
  - crystal
- DDS implementation:
  - Opensplice
- Client library (if applicable):
  - rclcpp_lifecycle

#### Steps to reproduce issue

```
ros2 run lifecycle lifecycle_talker
```

#### Expected behavior

lifecycle talker should be launch

#### Actual behavior
The terminal prints:

```
========================================================================================
Context     : DDS::OpenSplice::TypeSupport::register_type
Date        : 2018-12-20T15:46:06+0100
Node        : erle-MS-B09011
Process     : lifecycle_talker <28982>
Thread      : main thread 7fab6f121f40
Internals   : TypeSupport.cpp/79/6.9.181126OSS///-1
----------------------------------------------------------------------------------------
Report      : Bad parameter: Could not register type 'lifecycle_msgs::srv::dds_::Sample_ChangeState_Request_'.
Internals   : DDS::OpenSplice::DomainParticipant::wlReq_load_type_support_meta_holder/DomainParticipant.cpp/2293/3/1545317166.349231055
----------------------------------------------------------------------------------------
Report      : Deserialize failed
Internals   : Type deserialize/sd_serializerXMLTypeinfo.c/2945/0/1545317166.349239402
----------------------------------------------------------------------------------------
Report      : c_metaBind failed for Struct <ChangeState_Request_> in scope <lifecycle_msgs::srv::dds_>.
Internals   : sd_deserXmlStructure/sd_serializerXMLTypeinfo.c/2365/0/1545317166.349244866
----------------------------------------------------------------------------------------
Report      : Redeclaration of 'lifecycle_msgs::srv::dds_::ChangeState_Request_' doesn't match existing declaration
Internals   : c_metaObject::c_metaBind/c_metabase.c/272/0/1545317166.349251332

>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:
'failed to create responder: lifecycle_msgs::srv::dds_::Sample_ChangeState_Request_TypeSupport.register_type: bad domain participant or type name parameter, at /home/erle/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_service.cpp:125, at /home/erle/ros2_ws/src/ros2/rcl/rcl/src/rcl/service.c:178'

with this new error message:

  'service handle is null, at /home/erle/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_service.cpp:188'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hansvanthag, 2019-01-03:
to me it looks like there's a conflicting definition of the ChangeState_Request type in the system as it complains that the type-to-be-registered doesn't match an existing declaration

clalancette, 2019-01-03:
Hm, interesting.  I just tried locally here, and I am not able to reproduce on my system.  It looks like I can launch the lifecycle demos either with Fast-RTPS or with Opensplice, no problems.

@ahcorde How did you install opensplice?  What version of Opensplice are you using?  And how are you setting up the environment to use opensplice?

ahcorde, 2019-01-07:
I think that this is happening because I'm mixing ros bouncy and crystal. I will check this today and I will let you know.

clalancette, 2019-01-07:
> I think that this is happening because I'm mixing ros bouncy and crystal. I will check this today and I will let you know.

That could definitely cause problems; Bouncy was using OpenSplice 6.7 while Crystal uses 6.9.  Let us know what you find.

ahcorde, 2019-01-09:
Confirmed. It's because I was mixing bouncy and crystal.

thanks for your support.

closed 2019-01-09 by ahcorde