ros2/rmw_opensplice #258: rcl test_graph fails on windows nightly

sloretz, 2019-01-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - source
- Version or commit hash:
  - master
- DDS implementation:
  - Opensplice
- Client library (if applicable):
  - rcl

Failing nightly: https://ci.ros2.org/job/nightly_win_extra_rmw_rel/238
First failed: https://ci.ros2.org/job/nightly_win_extra_rmw_rel/190/#showFailuresLink

I've only seen this failure on the windows nightly. It looks like this has been failing since ros2/rmw_opensplice#251 was merged.

<details><summary>test_graph__rmw_opensplice_cpp output</summary>

```
      Start 58: test_graph__rmw_opensplice_cpp

58: Test command: C:\Python37\python.exe "-u" "C:/J/workspace/nightly_win_extra_rmw_rel/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/test_results/rcl/test_graph__rmw_opensplice_cpp.gtest.xml" "--package-name" "rcl" "--output-file" "C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/ament_cmake_gtest/test_graph__rmw_opensplice_cpp.txt" "--env" "RMW_IMPLEMENTATION=rmw_opensplice_cpp" "--append-env" "PATH=C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/Release" "--command" "C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/test/Release/test_graph__rmw_opensplice_cpp.exe" "--gtest_output=xml:C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/test_results/rcl/test_graph__rmw_opensplice_cpp.gtest.xml"
58: Test timeout computed to be: 60
58: -- run_test.py: extra environment variables:
58:  - RMW_IMPLEMENTATION=rmw_opensplice_cpp
58: -- run_test.py: extra environment variables to append:
58:  - PATH+=C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/Release
58: -- run_test.py: invoking following command in 'C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl':
58:  - C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/test/Release/test_graph__rmw_opensplice_cpp.exe --gtest_output=xml:C:/J/workspace/nightly_win_extra_rmw_rel/ws/build/rcl/test_results/rcl/test_graph__rmw_opensplice_cpp.gtest.xml
58: Running main() from gtest_main.cc
58: [==========] Running 9 tests from 2 test cases.
58: [----------] Global test environment set-up.
58: [----------] 6 tests from TestGraphFixture__rmw_opensplice_cpp
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_get_and_destroy_topic_names_and_types
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_get_and_destroy_topic_names_and_types (1350 ms)
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_count_publishers
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_count_publishers (1344 ms)
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_count_subscribers
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_count_subscribers (1343 ms)
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_graph_query_functions
58: [INFO] [rcl]:  Try 1: 0 publishers, 0 subscribers, and that the topic is not in the graph.
58: [INFO] [rcl]:   state correct!
58: [INFO] [rcl]:  Try 1: 0 publishers, 0 subscribers, and that the topic is not in the graph.
58: [INFO] [rcl]:   state wrong, waiting up to '400000000' nanoseconds for graph changes... 
58: [INFO] [rcl]: change occurred
58: [INFO] [rcl]:  Try 2: 0 publishers, 0 subscribers, and that the topic is not in the graph.
58: [INFO] [rcl]:   state wrong, waiting up to '400000000' nanoseconds for graph changes... 
58: [INFO] [rcl]: change occurred
58: [INFO] [rcl]:  Try 3: 1 publishers, 0 subscribers, and that the topic is in the graph.
58: [INFO] [rcl]:   state correct!
58: [INFO] [rcl]:  Try 1: 1 publishers, 0 subscribers, and that the topic is in the graph.
58: [INFO] [rcl]:   state wrong, waiting up to '400000000' nanoseconds for graph changes... 
58: [INFO] [rcl]: change occurred
58: [INFO] [rcl]:  Try 2: 1 publishers, 1 subscribers, and that the topic is in the graph.
58: [INFO] [rcl]:   state correct!
58: [INFO] [rcl]:  Try 1: 0 publishers, 1 subscribers, and that the topic is in the graph.
58: [INFO] [rcl]:   state correct!
58: [INFO] [rcl]:  Try 1: 0 publishers, 0 subscribers, and that the topic is not in the graph.
58: [INFO] [rcl]:   state correct!
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_graph_query_functions (1354 ms)
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_graph_guard_condition_topics
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '400000000' nanoseconds for graph changes
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_graph_guard_condition_topics (2556 ms)
58: [ RUN      ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_service_server_is_available
58: [INFO] [rcl]: waiting up to '1000000000' nanoseconds for graph changes
58: [INFO] [rcl]: waiting up to '1000000000' nanoseconds for graph changes
58: [       OK ] TestGraphFixture__rmw_opensplice_cpp.test_rcl_service_server_is_available (1351 ms)
58: [----------] 6 tests from TestGraphFixture__rmw_opensplice_cpp (9299 ms total)
58: 
58: [----------] 3 tests from NodeGraphMultiNodeFixture
58: [ RUN      ] NodeGraphMultiNodeFixture.test_node_info_subscriptions
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: [  FAILED  ] NodeGraphMultiNodeFixture.test_node_info_subscriptions (25228 ms)
58: [ RUN      ] NodeGraphMultiNodeFixture.test_node_info_publishers
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: C:\J\workspace\nightly_win_extra_rmw_rel\ws\src\ros2\rcl\rcl\test\rcl\test_graph.cpp(366): error: Expected equality of these values:
58:   num_topics
58:     Which is: 1
58:   nat.names.size
58:     Which is: 0
58: [  FAILED  ] NodeGraphMultiNodeFixture.test_node_info_publishers (17606 ms)
58: [ RUN      ] NodeGraphMultiNodeFixture.test_node_info_services
```
</details>

cwyark, 2019-06-21:
Is this issue solved ? It seems test_graph__rmw_opensplice_cpp test is passed in the latest nightly build.

supercjb, 2019-06-21:
it seems being solved:
https://ci.ros2.org/job/nightly_win_extra_rmw_rel/243/
please update the status.

clalancette, 2019-06-21:
Yes, we haven't seen this in a while.  I'm going to close this out.  Thanks for the heads up.

closed 2019-06-21 by clalancette