ros2/rmw_opensplice #259: Names conflict in message creation

alsora, 2019-01-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - debian packages
- Version or commit hash:
  - crystal
- DDS implementation:
  - OpenSplice

#### Steps to reproduce issue
I have a package defining custom interfaces.
It contains two interfaces:

- FooStamped.msg
`std_msgs/Header header`

- Header.msg
`int32 bar`

The package depends on `std_msgs` and includes it as dependency for creating the interfaces.

#### Behavior

Compilation goes fine with FastRTPS or Connext.

Compilation fails with Opensplice.

In file included from /root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/ccpp_FooStamped_.h:4:0,
                 from /root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/foo_stamped__rosidl_typesupport_opensplice_cpp.hpp:9,
                 from /root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp:4:
/root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/FooStamped_.h:39:33: error: ‘dds_’ in namespace ‘std_msgs::msg’ does not name a type
                ::std_msgs::msg::dds_::Header_ header_;
                                 ^~~~
/root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp: In function ‘void simple_interfaces::msg::typesupport_opensplice_cpp::convert_ros_message_to_dds(const __ros_msg_type&, simple_interfaces::msg::typesupport_opensplice_cpp::__dds_msg_type&)’:
/root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp:101:37: error: ‘using __dds_msg_type = struct simple_interfaces::msg::dds_::FooStamped_ {aka struct simple_interfaces::msg::dds_::FooStamped_}’ has no member named ‘header_’
     ros_message.header, dds_message.header_);
                                     ^~~~~~~
/root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp: In function ‘void simple_interfaces::msg::typesupport_opensplice_cpp::convert_dds_message_to_ros(const __dds_msg_type&, simple_interfaces::msg::typesupport_opensplice_cpp::__ros_msg_type&)’:
/root/msg_ws/build/simple_interfaces/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp:162:17: error: ‘const __dds_msg_type {aka const struct simple_interfaces::msg::dds_::FooStamped_}’ has no member named ‘header_’
     dds_message.header_, ros_message.header);
                 ^~~~~~~
make[2]: *** [CMakeFiles/simple_interfaces__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/simple_interfaces/msg/dds_opensplice/foo_stamped__type_support.cpp.o] Error 1
make[1]: *** [CMakeFiles/simple_interfaces__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2

alsora, 2019-01-25:
[dds_test.tar.gz](https://github.com/ros2/rmw_opensplice/files/2795643/dds_test.tar.gz)

Here it is. Tested with Bouncy and Crystal.
Build succeeds only if opensplice is not installed.

dirk-thomas, 2019-01-25:
The `idlpp` program generates C++ code with a non-namespaced include:

> #include "Header_.h"

even though the `.idl` file contains a namespaced include:

> #include "std_msgs/msg/dds_opensplice/Header_.idl"

Since all possible paths are part of the include directories the two `Header_.h` files "collide" and you get the one found in the first include directory.

I am not sure if we are using `idlpp` wrong here. Maybe @eboasson has an idea how to make this work in OpenSlice?

eboasson, 2019-01-29:
@dirk-thomas, @alsora: I know rather little about the inner workings of ``idlpp``, but I have found someone who does know it and who is now looking into it. That gives some hope for a quick resolution, or at least a workaround that is a little bit more elegant than editing the output of ``idlpp`` with ``perl`` (or whatever device one prefers) ...

eboasson, 2019-01-30:
I've since learnt that the problem is reasonably well understood and that the issue has been accepted as a bug in ``idlpp``. That means it will get fixed at some point, but unfortunately it also means that there is no schedule yet. We can always try and see if there is a way to manipulate the priorities 😉 — but such things are best reserved for the cases where there really is no other way. Can you give us an indication of the urgency?

As an alternative ... would my "maximally inelegant" workaround do the trick? I suspect that simply including both files named "Header_.h" whenever a file generated by ``idlpp`` contains an include directive for "Header_.h", then a bit of munging would do. Just match lines with that include directive, and in its stead output two lines for the two files. Something like:

```sh
perl -i -pe '$_ = "#include \"abc/Header_.h\"\n#include \"def/Header_.h\"\n" if /#include\s+"Header_\.h"/' xxx.cpp
```

does that, editing xxx.cpp in place. The only thing left is then to figure out which files to edit ...

dirk-thomas, 2019-01-30:
> Can you give us an indication of the urgency?

Imo fixing this issue is of pretty high priority. Atm namespace is basically broken in `idlpp`. Since ROS is developed in a federated way it is very likely that names within different packages will collide.

> As an alternative ... would my "maximally inelegant" workaround do the trick?

In this case I would think the user can workaround the problem by changing the name of it's own header message.

I don't think that modifying the generated code in a generic way is feasible. And custom patching is for case-to-case doesn't sound reasonable either.

If anyone would like to try a pull request to ` rosidl_typesupport_opensplice` would be an option.

eboasson, 2019-01-30:
@dirk-thomas, I wasn't trying to suggest putting such an ugly hack in the RMW implementation — that's an obviously bad idea. But as it seems this issue with ``idlpp`` has been around from the very begginning of OpenSplice I assumed the incidence of the problem to be rare. But this

> Since ROS is developed in a federated way it is very likely that names within different packages will collide.

is a good point. I'll see what I can do to get it expedited.

eboasson, 2019-02-06:
@dirk-thomas, word is that there is a fix making its way through the process. What I heard is that it will require passing an extra option to ``idlpp`` to avoid the risk of changing behaviour for current users. If you'd like to try it, I'll go and find out whether we could supply it as a patch to the OpenSplice CE prior to actually releasing it.

dirk-thomas, 2019-02-06:
> If you'd like to try it, I'll go and find out whether we could supply it as a patch to the OpenSplice CE prior to actually releasing it.

That would be great.

MarcelJordense, 2019-02-07:
@dirk-thomas Instead of a patch I have pushed a preliminary solution for this problem to my own fork of OpenSplice CE (https://github.com/MarcelJordense/opensplice.git) on branch ''maintain-include-namespace". Note that this solution is still to be approved by our internal process. 
To enable the idlpp preprocessor to maintain the include paths in the generated code the option "-o maintain-include-namespace" has to be provided.

dirk-thomas, 2019-02-11:
Can you maybe create a PR for adding the argument to the [idlpp invocation](https://github.com/ros2/rosidl_typesupport_opensplice/blob/87e30e8371fac2cab49c5fc249e9b9b4c5bb0bd5/rosidl_typesupport_opensplice_cpp/rosidl_typesupport_opensplice_cpp/__init__.py#L61-L77). If possible it would be great to make this change backward compatible - so only for versions of `idlpp` which support this argument actually pass it and for older versions which don't omit it.

That would allow us to release this fix into the latest ROS distro Crystal. Otherwise it could only be used in the next release.

MarcelJordense, 2019-02-26:
@dirk-thomas  A new release of OpenSplice CE (OSPL_V6_9_190221OSS_RELEASE) is available which contains the option to maintain the include path in the generated files.

dirk-thomas, 2019-02-26:
@MarcelJordense We just build a new Debian package based on that release: https://build.osrfoundation.org/view/All/job/opensplice-debbuilder_no_upload/8/

Unfortunately when trying to use the new command line option (added in ros2/rosidl_typesupport_opensplice#20) the `idlpp` invocation segfaults immediately without any output.

MarcelJordense, 2019-02-27:
@dirk-thomas The indicated problem should be solved in the OpenSplice CE release OSPL_V6_9_190227OSS_RELEASE

dirk-thomas, 2019-02-27:
The new Debian package seems to work fine: https://build.osrfoundation.org/view/All/job/opensplice-debbuilder_no_upload/9/

Will role new ones for all targets...

dirk-thomas, 2019-02-28:
I just imported the new Debian packages: http://build.ros2.org/job/import_upstream/126/

dirk-thomas, 2019-03-01:
I am not sure if I am missing something but in the latest builds (e.g. https://ci.ros2.org/job/ci_linux/6298/consoleFull#console-section-244) it seems that the code generated by `idlpp` even when using the new option `-o maintain-include-namespace` still generates include directives without the namespace:

```
.../build/lifecycle_msgs/rosidl_typesupport_opensplice_cpp/lifecycle_msgs/srv/dds_opensplice/GetAvailableTransitions_Response_.h:15:10: fatal error: TransitionDescription_.h: No such file or directory
  #include "TransitionDescription_.h"
```

Can you please double check the behavior on your end.

dirk-thomas, 2019-03-01:
Sometimes `idlpp` only generates a subset of the files and returns with the error code 255. The only output it prints is `Error: running cppgen.` which doesn't give any information why it failed. Also when re-running the very same command `idlpp` passes just fine and generates all files as expected. Are there any option to debug `idlpp` and get more verbose information?

In other cases where the input `.idl` file contains two include directives with namespace the generated `<idlname>.h` contains one include directive which lack the namespace while the second one does.

MarcelJordense, 2019-03-01:
@dirk-thomas I will look into it.

MarcelJordense, 2019-03-05:
@dirk-thomas It appeared that the maintain-include-namespace option did not correctly worked when an include file included another include file. The OSPL_V6_9_190227OSS_RELEASE was updated which should fix this issue.

dirk-thomas, 2019-03-05:
> The OSPL_V6_9_190227OSS_RELEASE was updated which should fix this issue.

It is very bad practice to change the content of a tagged release after the fact and not give any indication whatsoever for a user to distinguish if they have the flawed version or the hopefully working newer version. I can only strongly suggest to create new tags / versions / releases in the future to avoid that.

We will roll new Debian packages and try to reapply the reverted patch...

dirk-thomas, 2019-03-06:
Unfortunately I am still getting a segfault with `free(): invalid pointer`: https://github.com/ADLINK-IST/opensplice/blob/ee3ea69231ee974f633331708cbb5e96bd9d2abb/src/tools/idlpp/code/idl_genSpliceDep.c#L153

MarcelJordense, 2019-03-06:
@dirk-thomas I tried to reproduce this problem by building ROS using this option but the problem did not occur. Do you have information about on which platform and on which the idl file the problem occurred?

dirk-thomas, 2019-03-06:
On Ubuntu 18.04 amd64 when building the `test_msgs` package processing `StaticArrayNested_.idl`. (I am not using the default branches but the ones in this PR: ros2/rosidl#346 but I wouldn't expect that to make a difference.)

dirk-thomas, 2019-03-12:
@MarcelJordense I have rebased ros2/rosidl_typesupport_opensplice#21 and ros2/rosidl_typesupport_opensplice#23 in the mean time since other significant changes have landed on master. The problem is still the same (see the latest set of builds on the first referenced PR). Where you able to reproduce the problem and come up with an updated patch?

MarcelJordense, 2019-03-13:
@dirk-thomas I was not able to reproduce the problem. I used the master branch of ROS2 and it seem to build successfully. I checked the generated files and they seem ok. I will try again with the latest master.

dirk-thomas, 2019-03-18:
Moving back to "in progress" until there is an update OpenSplice release.

MarcelJordense, 2019-03-26:
@dirk-thomas There is now a new release of OpenSplice CE available (OSPL_V6_9_190321OSS_RELEASE) which hopefully solves the problem.

dirk-thomas, 2019-03-26:
I just tried the latest version on macOS and Windows and surprisingly the results are inconsistent:

* The [macOS build](https://ci.ros2.org/job/ci_osx/5349/console#console-section-14) failed and it looks like some of the generated include directives are still not using the namespace / full path:

  ```
  build/lifecycle_msgs/rosidl_typesupport_opensplice_cpp/lifecycle_msgs/srv/dds_opensplice/GetAvailableStates_SplDcps.h:12:10: fatal error: 'State_SplDcps.h' file not found
  #include "State_SplDcps.h"
  ```

* The [Windows build](https://ci.ros2.org/job/ci_windows/6325/) on the other hand passed. And the generated code of the very same file contains a full path in the include directive:

  ```
  #include "lifecycle_msgs/msg/dds_opensplice/State_SplDcps.h"
  ```

MarcelJordense, 2019-04-03:
@dirk-thomas It seems that the problem was caused when determining the dependency of a sequence type where the subtype is located in another file. I have made a new release which should solve this problem (OSPL_V6_9_190403OSS_RELEASE).

dirk-thomas, 2019-04-03:
The latest macOS build with ros2/rosidl_typesupport_opensplice#21 passes now: [image]

I will look into deploying the latest release to all Jenkins nodes as well as roll a Debian package...

dirk-thomas, 2019-04-04:
The problem has been addressed in OpenSplice 6.9.190403. You will also need the latest state of [rosidl_typesupport_opensplice](https://github.com/ros2/rosidl_typesupport_opensplice/pull/23).

@MarcelJordense Thank you for addressing the problem.

closed 2019-04-04 by dirk-thomas