ros2/rmw_opensplice #279: Compile error for version 0.7.2 (dashing release) on Ubuntu 16.04 and GCC5: event_compiler.cpp no match for call [...]

MatteoRagni, 2019-07-26:
## Bug report

- **Operating System**: Ubuntu 16.04 with PREEMPT_RT patch
- **Installation type**: Compilation with `colcon build --merge-install` of the `latest-release` from sources, `libopensplice69` installed via package manager (ROS2 repos). 
- **Compiler**: GCC5
- **Version or commit hash**: `0.7.2` in `rmw_opensplice_cpp/packages.xml`, 

#### Steps to reproduce issue

During a compilation from sources of a ROS2 distribution (actual `latest-release`, wich is a `dashing` based) thc compilation stops with error at package `rmw_implementation_cpp` with the error reported below.

```
Starting >>> rmw_opensplice_cpp
--- stderr: rmw_opensplice_cpp                             
In file included from /usr/include/c++/5/bits/hashtable.h:35:0,
                 from /usr/include/c++/5/unordered_map:47,
                 from /home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:20:
/usr/include/c++/5/bits/hashtable_policy.h: In instantiation of ‘struct std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> >’:
/usr/include/c++/5/type_traits:137:12:   required from ‘struct std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > >’
/usr/include/c++/5/type_traits:148:38:   required from ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
/usr/include/c++/5/bits/unordered_map.h:100:66:   required from ‘class std::unordered_map<rmw_event_type_t, unsigned int>’
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:23:68:   required from here
/usr/include/c++/5/bits/hashtable_policy.h:85:34: error: no match for call to ‘(const std::hash<rmw_event_type_t>) (const rmw_event_type_t&)’
  noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
                                  ^
In file included from /usr/include/c++/5/bits/move.h:57:0,
                 from /usr/include/c++/5/bits/stl_pair.h:59,
                 from /usr/include/c++/5/utility:70,
                 from /usr/include/c++/5/unordered_map:38,
                 from /home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:20:
/usr/include/c++/5/type_traits: In instantiation of ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’:
/usr/include/c++/5/bits/unordered_map.h:100:66:   required from ‘class std::unordered_map<rmw_event_type_t, unsigned int>’
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:23:68:   required from here
/usr/include/c++/5/type_traits:148:38: error: ‘value’ is not a member of ‘std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > >’
     : public integral_constant<bool, !_Pp::value>
                                      ^
In file included from /usr/include/c++/5/unordered_map:48:0,
                 from /home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:20:
/usr/include/c++/5/bits/unordered_map.h: In instantiation of ‘class std::unordered_map<rmw_event_type_t, unsigned int>’:
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:23:68:   required from here
/usr/include/c++/5/bits/unordered_map.h:100:66: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef __umap_hashtable<_Key, _Tp, _Hash, _Pred, _Alloc>  _Hashtable;
                                                                  ^
/usr/include/c++/5/bits/unordered_map.h:107:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::key_type key_type;
                                             ^
/usr/include/c++/5/bits/unordered_map.h:108:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::value_type value_type;
                                               ^
/usr/include/c++/5/bits/unordered_map.h:109:48: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::mapped_type mapped_type;
                                                ^
/usr/include/c++/5/bits/unordered_map.h:110:43: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::hasher hasher;
                                           ^
/usr/include/c++/5/bits/unordered_map.h:111:46: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::key_equal key_equal;
                                              ^
/usr/include/c++/5/bits/unordered_map.h:112:51: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::allocator_type allocator_type;
                                                   ^
/usr/include/c++/5/bits/unordered_map.h:117:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::pointer  pointer;
                                             ^
/usr/include/c++/5/bits/unordered_map.h:118:50: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::const_pointer const_pointer;
                                                  ^
/usr/include/c++/5/bits/unordered_map.h:119:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<rmw_event_type_t> >, std::__detail::__is_noexcept_hash<rmw_event_type_t, std::hash<rmw_event_type_t> > > >’
       typedef typename _Hashtable::reference  reference;
                                               ^
[... 41 lines cut ...]
       ^
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:28:1: error: too many initializers for ‘const std::unordered_map<rmw_event_type_t, unsigned int>’
 };
 ^
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp: In function ‘DDS::StatusKind get_status_kind_from_rmw(rmw_event_type_t)’:
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:32:19: error: ‘const class std::unordered_map<rmw_event_type_t, unsigned int>’ has no member named ‘at’
   return mask_map.at(event_t);
                   ^
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp: In function ‘bool is_event_supported(rmw_event_type_t)’:
/home/user/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/event_converter.cpp:37:19: error: ‘const class std::unordered_map<rmw_event_type_t, unsigned int>’ has no member named ‘count’
   return mask_map.count(event_t) > 0;
                   ^
make[2]: *** [CMakeFiles/rmw_opensplice_cpp.dir/src/event_converter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/rmw_opensplice_cpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rmw_opensplice_cpp	[ Exited with code 2 ]
```

A lot of stuff that goes into `c++` headers, I hope it is not a compiler's libraries error.

#### Additional information

I know that dashing does not actually support 16.04, but I have constraints on software versions.

clalancette, 2019-07-26:
I suspect that it does have to do with the compiler version.  What happens if you uninstall `libopensplice69`, then build from scratch?  Do you get further?  I suspect there will be other trouble, but I haven't tried on 16.04 in quite a while.

In general, we won't have time to look into problems on unsupported platforms.  However, if you are able to find and fix the problems, we are happy to review patches that make things more backwards compatible.  Good luck!

MatteoRagni, 2019-07-26:
I'm preparing a virtual machine with a clone of the system for compilation testing.

MatteoRagni, 2019-07-26:
@clalancette without `libopensplice69` it has gone further. I'm still waiting for colcon to finish.

MatteoRagni, 2019-07-29:
With GCC7, it compiles without isseus, with opensplice installed

closed 2019-07-29 by MatteoRagni