ros2/rmw_opensplice #282: OpenSplice exhibits bad performance with ROS 2 Navigation Stack

rotu, 2019-09-11:
## Bug report

(Moved here from https://github.com/ros-planning/navigation2/issues/1094. See that issue for slightly more discussion.)

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-dashing-nav2-amcl/bionic,now 0.2.3-1bionic.20190831.172054 amd64 [installed]
- DDS implementation:
  - OpenSplice (with changes to ros_ospl.xml to set AllowMulticast spdp)
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Using the launch files here: https://github.com/RoverRobotics/openrover-ros2/tree/98fa2973b726bc8b867ce1980b4063f210585584/openrover_demo/launch
```
# on the rover
ros2 launch openrover_demo drive.launch.py
# on a computer
ros2 launch openrover_demo rviz.launch.py
# on either the computer or the rover
ros2 launch openrover_demo amcl.launch.py
```
give an initial pose in RViz with the 2D Pose Estimate tool.

#### Expected behavior
AMCL will publish the location of the robot or display an error message why it cannot.

#### Actual behavior
Occasionally, AMCL will start and I will see the robot location and lidar data in RViz within a few seconds.

Most of the time AMCL acknowledges the initial pose and then does not publish a tf for a long time (minutes), and does not indicate any status information.

Here is some example output from the Nav2 stack:
```
[amcl-3] [INFO] [amcl]: Creating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting the system bringup...
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating map_server
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[map_server-2] [WARN] [map_server]: Mode parameter not set, using default value (trinary): bad conversion
[map_server-2] [INFO] [map_server]: Activating
[map_server-2] [INFO] [map_server]: OccGridLoader: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring and activating amcl
[amcl-3] [INFO] [amcl]: Configuring
[amcl-3] [INFO] [amcl]: initTransforms
[amcl-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[amcl-3] [INFO] [amcl]: initPubSub
[amcl-3] [INFO] [amcl]: Subscribed to map topic.
[amcl-3] [INFO] [amcl]: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: The system is active
[amcl-3] [INFO] [amcl]: initialPoseReceived
[amcl-3] [INFO] [amcl]: Setting pose (1567563024.080354): -0.313 -2.710 -2.384
[amcl-3] [INFO] [amcl]: Received a 567 X 329 map @ 0.050 m/pix
[amcl-3] 
[amcl-3] [INFO] [amcl]: Setting pose (1567563025.430725): -0.313 -2.710 -2.384
[amcl-3] [INFO] [amcl]: initialPoseReceived
[amcl-3] [INFO] [amcl]: Setting pose (1567563033.278710): -0.107 -1.923 -2.367
```

a minute and a half later:
```
[amcl-3] [INFO] [amcl_rclcpp_node]: [signalFailure] Drop message: frame 'lidar_link' at time 1567563119.337 for reason(0)
[... 58 lines cut ...]

> I have had a quick look @rotu ... it ain't pretty. A really detailed study would take more time, but I'm not sure that would really make a difference in practice. What I see:
> 
>     * 7 processes on the one node, 11 on the other.
> 
>     * Each process tries to learn the topic definitions, readers (+ DCPS subscribers), writers (+ DCPS publishers) of all other processes. Without multicast (i.e., as you described) other than for initial participant discovery, with 18 processes that's necessarily quite a bit of data.
> 
>     * The vast majority of the traffic is topic discovery. It seems plausible that the XML parsing & validation by the "spliced" service can't keep up, and as it is the "dq.builtin" thread that dumps this data into a data reader for further processing by that service, this fits with the earlier observation concering the "number of samples surpassing the warning level".
> 
>     * To make matters worse, there are quite a few retransmits that appear to be caused by packet loss. My guess is that with up to 17 processes sending to one at the same time, the UDP receive buffers can easily overflow, leading to packet loss and (even more) retransmits.
> 
>     * If the peers are all known when a topic/reader/writer is created, the corresponding discovery data is sent straight to all peers (using multicast if that's allowed). If they are not yet known, they'll request retransmits later, which is even more expensive.
> 
> 
> Given the numbers of processes on the two nodes and the dominant traffic being topic discovery, one would naively expect a 11/7 ratio in traffic for the two machines. Wireshark reports 877474 units for 10.0.2.22 and 601316 for 10.0.0.8. That is pretty close, so I think nothing more than this is going on.
> 
> What to do:
> 
>     * Use shared memory, if you can get your hands on a shared memory-enabled build. When you do that, suddenly there are only two nodes that need to exchange topic definitions, rather than 18 processes. Obviously going from 18 nodes all-to-all to 2 nodes all-to-all is a gigantic reduction in traffic.
> 
>     * A hack to disable topic discovery. It'll run just fine without it as long as you don't need transient/persistent data. That's literally a hack in the DDSI service source code, but a really small one (though I have done it before, I'd have to dig into the sources to figure it out again).
> 
>     * Enlarge the UDP socket receive buffers. There is an option for that (Internal/MinimumSocketReceiveBufferSize) but it would probably also require raising the kernel's limit on the socket receive buffer size. I don't know how much it would help, but I do suspect it'd help a bit.
> 
>     * ... just use Cyclone DDS grin

rotu, 2019-09-17:
@eboasson I just tried with shared memory enabled on OpenSplice 6.10.2p4, and the results are encouraging: bringup is about 10s.

On the other hand, even with shared memory *disabled*, I'm still seeing pretty reasonable performance on 6.10.2p4 (15 seconds), which leads me to believe that simply using OpenSplice Community 6.10 instead of 6.9 *may* resolve this performance issue.

dirk-thomas, 2019-09-17:
The latest public release of OpenSplice OSS I see is 6.9 from 2019-07-05: https://github.com/ADLINK-IST/opensplice/releases

rotu, 2019-09-17:
@dirk-thomas You're correct - it's not yet released. @joespeed says it's in the works. My testing on 6.10.2p4 is based on the commercial distribution.

dirk-thomas, 2020-01-03:
> @joespeed says it's in the works.

Any update on this?