ros2/rmw_opensplice #291: Console output when running composition demo with OpenSplice

sloretz, 2019-10-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu bionic
- Installation type:
  - Binaries
- Version or commit hash:
  -
   ```
   ros-eloquent-composition/bionic,bionic,now 0.8.1-1bionic.20191024.073050 amd64 [installed,automatic]
   ros-eloquent-composition-interfaces/bionic,bionic,now 0.8.0-1bionic.20191024.065600 amd64 [installed,automatic]
   ros-eloquent-examples-rclcpp-minimal-composition/bionic,bionic,now 0.8.1-1bionic.20191024.073050 amd64 [installed,automatic]
   ros-eloquent-opensplice-cmake-module/bionic,bionic,now 0.8.1-1bionic.20191024.044026 amd64 [installed,automatic]
   ros-eloquent-rmw-opensplice-cpp/bionic,bionic,now 0.8.1-1bionic.20191024.062945 amd64 [installed]
   ros-eloquent-rosidl-typesupport-opensplice-c/bionic,bionic,now 0.8.1-1bionic.20191024.062611 amd64 [installed,automatic]
   ros-eloquent-rosidl-typesupport-opensplice-cpp/bionic,bionic,now 0.8.1-1bionic.20191024.062531 amd64 [installed,automatic]
   ```
- DDS implementation:
  - Opensplice
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Following https://index.ros.org/doc/ros2/Tutorials/Composition/

In one terminal
```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run rclcpp_components component_container
```

In another terminal
```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Talker
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Listener
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component list
```

#### Expected behavior
I'd expect terminal 2 output to be

```bash
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Talker
Loaded component 1 into '/ComponentManager' container node as '/talker'
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Listener
Loaded component 2 into '/ComponentManager' container node as '/listener'
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 
/ComponentManager
  1  /talker
  2  /listener
```
#### Actual behavior

There is much more console output

```bash
sloretz@sloretz-XPS-15-9560:~$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Talker
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:17-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16656>
Thread      : main thread 7fed712ae740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
Loaded component 1 into '/ComponentManager' container node as '/talker'
sloretz@sloretz-XPS-15-9560:~$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component load /ComponentManager composition composition::Listener
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:22-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16689>
Thread      : main thread 7fc0475ea740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:22-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16689>
Thread      : main thread 7fc0475ea740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
Loaded component 2 into '/ComponentManager' container node as '/listener'
sloretz@sloretz-XPS-15-9560:~$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 component list
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:26-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16722>
Thread      : main thread 7fc7dd361740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:26-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16722>
Thread      : main thread 7fc7dd361740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-10-25T14:16:26-0700
Node        : sloretz-XPS-15-9560
Process     : python3.6 <16722>
Thread      : main thread 7fc7dd361740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
/ComponentManager
  1  /talker
  2  /listener
```

#### Additional information
This does not happen with fastrtps or connext

sloretz, 2019-11-13:
Grouping this since I assume the cause is the same, there is also a lot of console output when using `ros2 node info` and opensplice

First terminal

```
ros2 run examples_rclcpp_minimal_composition composition_composed
```

Second terminal

```
ros2 node info /publisher_node
```

Expected output
```
/publisher_node
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
    /topic: std_msgs/msg/String
  Services:
    /publisher_node/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /publisher_node/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /publisher_node/get_parameters: rcl_interfaces/srv/GetParameters
    /publisher_node/list_parameters: rcl_interfaces/srv/ListParameters
    /publisher_node/set_parameters: rcl_interfaces/srv/SetParameters
    /publisher_node/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Action Servers:

  Action Clients:

```

Actual output
```
========================================================================================
Report      : WARNING
Date        : 2019-11-13T13:36:49-0800
Description : using network interface wlp2s0 (172.23.3.183) selected arbitrarily from: wlp2s0, docker0

Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : ddsi2 7f146a801700
Internals   : 6.9.190705OSS///ddsi2/q_nwif.c/989/0/1573681009.710927737/101
/publisher_node
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-11-13T13:36:51-0800
Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : main thread 7f147c4ee740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-11-13T13:36:51-0800
Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : main thread 7f147c4ee740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
    /topic: std_msgs/msg/String
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-11-13T13:36:51-0800
Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : main thread 7f147c4ee740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
  Services:
    /publisher_node/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /publisher_node/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /publisher_node/get_parameters: rcl_interfaces/srv/GetParameters
    /publisher_node/list_parameters: rcl_interfaces/srv/ListParameters
    /publisher_node/set_parameters: rcl_interfaces/srv/SetParameters
    /publisher_node/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-11-13T13:36:51-0800
Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : main thread 7f147c4ee740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
  Action Servers:

========================================================================================
Report      : NONE
Context     : DDS::OpenSplice::DataReader::wait_for_historical_data
Date        : 2019-11-13T13:36:51-0800
Node        : sloretz-XPS-15-9560
Process     : python3.6 <28565>
Thread      : main thread 7f147c4ee740
Internals   : DataReader.cpp/855/6.9.190705OSS///101
  Action Clients:

```

dirk-thomas, 2019-12-12:
@ThijsSassen Can you comment on this?

martinbremmer, 2019-12-13:
Quick analysis.
DDS::DataReader::wait_for_historical_data() is called in rmw_opensplice.
However, that returns ```DDS::RETCODE_PRECONDITION_NOT_MET```.
This'll happen when OpenSplice instance doesn't support durability features. Neither Durability service nor DuribilityClient are enabled within the configuration xml. So, the return code is by design.

A return that is not ```DDS::RETCODE_OK``` nor ```DDS::RETCODE_TIMEOUT``` triggers a report.
The report in this case, however, is not very helpful and maybe we should fix that in OpenSplice.

wjwwood, 2020-01-02:
Should this be closed?

ivanpauno, 2020-01-16:
Closing, as it's a bug in OpenSplice.

closed 2020-01-16 by ivanpauno