ros2/rmw_opensplice #292: MANUAL_BY_TOPIC Liveliness QoS does not seem to be working

hidmic, 2019-11-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, on both AMD64 and ARM64
- Installation type:
  - Binaries and from source
- Version or commit hash:
  - https://github.com/ros2/rmw_opensplice/commit/1213249ca15e38225b374ddcacc3e03769f0ec78

#### Steps to reproduce issue
Run the interactive QoS demo:

1. On one terminal:
```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run quality_of_service_demo_cpp interactive_publisher --delay 0.5 --deadline 1 --liveliness MANUAL_BY_TOPIC --lease 1
```

2. On another terminal:
```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run quality_of_service_demo_cpp interactive_subscriber --deadline 1 --liveliness MANUAL_BY_TOPIC --lease 1
```

Then, press `s` on the publisher's terminal to stop / start publishing.

#### Expected behavior

Liveliness change events are logged by the subscriber as publications stop / start.

#### Actual behavior

Liveliness changes when the publisher starts and only changes again when it gets killed (as if it were MANUAL_BY_NODE liveliness).

hidmic, 2019-11-13:
Curiously enough, the [liveliness demo](https://github.com/ros2/demos/tree/master/quality_of_service_demo#liveliness) does work. In that demo both talker and listener are in the same process.

dirk-thomas, 2019-12-05:
@ThijsSassen I am not sure if you are the right person to poke from ADLINK but I think this needs someone from ADLINK to look into and propose a patch. Please feel free to point me to a better person of contact.

ThijsSassen, 2019-12-06:
@dirk-thomas we will take a look at it.

ThijsSassen, 2019-12-09:
@dirk-thomas how important is this functionality. Currently OpenSplice DDSI does not support this and in Cyclone DDS we just implemented this functionality. We could backport it to OpenSplice but this requires work that we currently don't have scheduled for OpenSplice.

@hidmic The main difference between the interactive demo and the liveliness demo is that the latter is a single federation, meaning that it is just handled by the kernel (no DDSI necessary)

dirk-thomas, 2019-12-09:
> how important is this functionality.

The demo is just that: a demo. If users want to use that functionality it might be very important to them. On the other hand it might not matter to other users.

@ThijsSassen It would be good to state that OpenSplice doesn't support it, e.g. in the README of this repo. If you don't plan to implement this feature can you create a PR for that?

dirk-thomas, 2019-12-12:
> It would be good to state that OpenSplice doesn't support it, e.g. in the README of this repo.

Added a note in #298.