ros2/rmw_opensplice #296: Localhost restriction enforced from rclcpp not implemented

AlexisTM, 2019-11-27:
## Bug report

There was no followup in the implementation https://github.com/ros2/rmw_opensplice/pull/288 of the `ROS_LOCALHOST_ONLY` environment variable. 

This became part of the API since Eloquent release. (see [Relase notes](https://discourse.ros.org/t/ros-2-eloquent-elusor-released/11624))

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source (eloquent release)
- Version or commit hash:
  - v0.8.1: 1213249ca15e38225b374ddcacc3e03769f0ec78
- DDS implementation:
  - `rmw_opensplice`
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce the issue

``` bash
export ROS_LOCALHOST_ONLY=1
export RMW_IMPLEMENTATION=rmw_opensplice_cpp
ros2 run examples_rclcpp_minimal_publisher publisher_lambda
```

#### Expected behavior

The node uses the localhost interface only.

#### Actual behavior

The node uses the interface configuration (in this case, not defined thus arbitrarily chosen).

```bash
$ export ROS_LOCALHOST_ONLY=1
$ export RMW_IMPLEMENTATION=rmw_opensplice_cpp
$ ros2 run examples_rclcpp_minimal_publisher publisher_lambda

========================================================================================
Report      : WARNING
Date        : 2019-11-27T09:47:54+0100
Description : using network interface wlp59s0 (192.168.3.29) selected arbitrarily from: wlp59s0, br-e0ceb0d676d6, br-17184af4af26, docker0, br-d5254965f28e, enx98e743a956bc

Node        : Oreo
Process     : publisher_lambda <10792>
Thread      : ddsi2 7f600c3ad700
Internals   : 6.9.190705OSS///ddsi2/q_nwif.c/989/0/1574844474.713621116/0
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 1'
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 2'
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 3'
```

dirk-thomas, 2019-11-27:
@eboasson can you implement this when you have a chance (see https://github.com/ros2/rmw_opensplice/pull/288#issuecomment-543843176)?

eboasson, 2019-11-28:
@dirk-thomas I won’t be able to do that very soon, but perhaps someone else might have a chance in the next week or so (@cwyark @MarcelJordense ?). It believe it is sufficient to modify the ospl.xml file to set ``General/NetworkInterfaceAddress`` to, e.g., ``${OSPL_NETWORK_INTERFACE:-auto}``, and then setting ``OSPL_NETWORK_INTERFACE`` to 127.0.0.1 in ``rmw_create_node`` if in localhost-only mode.

Obviously this won’t have the desired effect if someone uses a custom ospl.xml file instead of the one included in the ROS2 install. But in that case, presumably the user is ok with configuring things, and so it doesn’t seem like a big problem to me.

martinbremmer, 2019-12-10:
The proposal of @eboasson works. I'll go ahead and create some pull request.