ros2/rmw_opensplice #299: Implement the rmw_get_publishers_info_by_topic and rmw_get_subscriptions_info_by_topic methods

jaisontj, 2019-12-11:
As a follow up to [this](https://github.com/ros2/rmw_opensplice/pull/289) PR which returns `RMW_RET_UNSUPPORTED` for the `rmw_get_publishers_info_by_topic` and `rmw_get_subscriptions_info_by_topic`; ensure that these methods are implemented. 

These methods are implemented in `rmw_fastrtps` in [this](https://github.com/ros2/rmw_fastrtps/pull/336) PR which can be used as a reference.

ivanpauno, 2019-12-11:
I finally realize that I will have to do this, as I'm moving all the graph API code to another package as part of https://github.com/ros2/rmw/issues/183. So, it doesn't worth it to address the task independently.

clalancette, 2021-10-07:
Closing since we no longer support OpenSplice in any active ROS 2 distribution.