ros2/rmw_opensplice #301: rmw opensplice implementation does not use asynchronous publishing

wjwwood, 2020-01-21:
See https://github.com/ros2/rmw_cyclonedds/issues/89 for most of the details, but the basics are:

As discussed elsewhere (links TBD), the current best practice is to use the closest equivalent to the asynchronous publishing mode that some DDS implementations have for implementing `rmw_publish()`. Right now, this repository does not change the QoS for this when creating a publisher and therefore I assume it uses something like the synchronous publishing mode.

So a few questions:

 - does OpenSplice have an "asynchronous publishing mode"-like setting?
 - if not, what is the blocking behavior of publishing in OpenSplice, or more generally what happens when you call publish?

If there is an equivalent mode then it would be best to switch to it by default, even if it has a negative performance impact for some cases.

We can/should separately decide if the best practice should be adjusted, but until that time we should try to keep our rmw implementations as consistent as possible.

We can also chose to not do anything about this, given that we're doing away with OpenSplice support in ROS.

eboasson, 2020-01-22:
OpenSplice only ever does "asynchronous" publishing.

wjwwood, 2020-01-23:
Thanks I see, so nothing needs to be done here. Thanks @eboasson, I'll close this.

If you have a pointer to OpenSplice docs about this, I'm sure someone (maybe future us) will appreciate it, but it's not required.

closed 2020-01-23 by wjwwood